EmuELEC/EmuELEC #286: Several possible bugs

JMPMX, 2020-09-03:
**EmuELEC version:**
 - Version 3.7

**Device**
 - SOC: S095
 - Brand: RKM (Rikomagic)
 - Model MK06 (gxbb_p200_1G_1Gbit_RealtekWiFi)

Good morning, I'd like to share the possible bugs I've found and possible improvements to be made:

1st in the following path: "/storage/.config/emuelec/configs/emuelec.conf"

# ------------ C - Audio ------------ #

## Set system volume (0..100)
audio.volume=10
It has no effect on the sound, neither system nor retouching. Possible bug?

2º The default system sound of both menus and videos is too loud by default (100%), curiously there is no option in the audio menu to change the system sound, as it appears in the Windows versions of Emulationstation (I use Retrobat), but after much research I was able to find a solution, modifying this parameter:

path: /storage/.config/emuelec/scripts

file "rr_audio.sh" change the value from "100" to "40" in the line:

RR_AUDIO_VOLUME="40".

I understand that if this part is not changed in the next versions it could be indicated in the Wiki for the rest of the users who have the same problem.

3º The sound of the emulator by default is too high, in this case is almost always Retroarch and can be configured to start a rom with hotkey+scratch up or hotkey+scratch down, I personally leave it at 10% and is more than enough.

I understand that this part if not changed in future versions could be indicated in the Wiki for other users who have the same problem.

4º The famous "Hotkey" button, I haven't found any information about it in the wiki and it is fundamental to be able to go out of retroarch among other things, this information could be included in the Wiki:

"Hotkey button": You can skip this task by holding down any button. If you have a dedicated button like the Xbox button or the PlayStation button, you can assign it so that some of the emulator shortcuts use it.

    Hotkey + Start = exit the emulator
    Hotkey + top right trigger (R1) = save status
    Hotkey + top left trigger (L1) = load saved status
    Hotkey + Left = decrease current saved status slot number
    Hotkey + Right = increase the slot number of the current saved status
    Hotkey + X = quick menu (with access to most of these other items)
    Hotkey + B = restart game
    Hotkey + Crosshead up = increase volume
    Hotkey + Crosshead down = Volume down

5º I use for consoles the romset No-Intro filtered 1G1R in such a way that leaves me a game with the preference EUR>USA>JAP, but when scraping roms in some systems like "Master System" the game covers are obtained from the Brazilian version, for example "Zillion (Europe, Brazil) (Rev 2).sms" should take the European version not the Brazilian one when scraping, possible bug.

6º Retroarch does not save the configurations of each core (path: /storage/.config/retroarch/config/"core folder"/"core name". opt) and this is a big problem since if we change in my case from the quick menu the options for for example Mupen-plus Next to increase the resolution of the games or the format from 4:3 to 16:9 or any other option has no effect on the games because the options of the .opt file are not overwritten, I don't know if it is a retroarch or linux problem because of writing permissions...

7º When you turn off Emuelec the TV-Box remains on because the Led is still on, it seems that the option to turn it off leaves it really in suspension, I don't know if it is a problem of EmulationStation or the Linux system that doesn't turn it off correctly.

Best regards,

A retrofan.

shantigilbert, 2020-09-04:
Thank you, I will try to go through each. 

All the audio related questions are a matter of taste, I find the volume is fine for my needs, and you are the very first person to even comment on this, so I am guessing it is also fine for most. So in case anyone needs a different volume then can either manage it on their TV or use your workaround? Sound mixing is not that same as in Windows so no real easy way to add volume control. 

You are right about the hotkeys, although they were in fact documented sometime probably got lost, will implement them in the WiKi, or, you can do it since its an open WiKi. 

5.- No idea, This is a screen scraper issue. 

6._ I am pretty sure it does save them as I use them all the time, so something else is going on on your side, if you can send the exact steps reproduce it from a clean image, I can take a look at it. 

7.- Depends on the TVbox

JMPMX, 2020-09-09:
"Points 1-2-3"

Regarding the issue of audio, maybe the rest does not have the problem because it connects to TV and lowers the volume with the control, I connect it to a monitor and it is possible that that is why the result is different, however in the Audio menu would not be wrong to include the volume bar that usually bring the ES compilations.

"No idea, This is a screen scraper issue"

I will investigate it and if I discover something I will expose it here.

"6.I am pretty sure it does save them as I use them all the time, so something else is going on on your side, if you can send the exact steps to play it from a clean image, I can take a look at it."

For some reason it didn't work for me the other day but today I tried it and as you say it works perfectly.

"7.- Depends on the TVbox"

I don't understand what the TVBox has to see, I understand that the shutdown function should not be executed correctly or is not correctly implemented for my TVBox (I don't know if it stops with other models or I have been the exception to the rule) and it only closes ES, while the Linux system is in suspension when it should close and leave the hardware without power to disconnect.

On the other hand I have seen that some common systems are missing like "CAVE" or "Coleccovision", I have tried to add it manually and although it shows you the theme in the menu the roms don't start because of an "override" problem according to the log, I have tried with the FBA core and MAME2003+. The code would be:
<system>
		<name>cave</name>
		<fullname>CAVE</fullname>
		<manufacturer>CAVE</manufacturer>
		<release>1994</release>
		<hardware>arcade</hardware>		
		<path>/storage/roms/cave</path>
		<extension>.zip .ZIP .7z .7Z</extension>
		<command>/emuelec/scripts/emuelecRunEmu.sh %ROM% -P%SYSTEM% --core=%CORE% --emulator=%EMULATOR% --controllers="%CONTROLLERSCONFIG%"</command>
		<platform>arcade</platform>
		<theme>cave</theme>
		<emulators>
			<emulator name="AdvanceMame">
				<cores>
					<core>AdvanceMame</core>
				</cores>
			</emulator>
            <emulator name="libretro">
                <cores>
                    <core>mame2003_plus</core>
                    <core>mame2010</core>
                    <core default="true">fbneo</core>
                    <core>fbalpha2012</core>
                </cores>
            </emulator>
        </emulators>
	</system>

Best regards,

A retrofan.

Cee123, 2020-09-26:
With the hotkeys, I've discovered a couple of things and it took me a while to figure out. But you probably already know this. But for anyone who doesn't. If you choose a hotkey button, then that button will need to be held before you press any of the other hotkeys that you've defined (i.e. fast forward, exit retroarch). And if you delete the hotkey, and you can do that in the config file under input_enable_hotkey = "button" you just set that to "nul" and then any of the hotkeys you've set (exit retroarch etc) only require a single button press. I think to exit retroarch would be the same button you choose to "restart retroarch". I think it's also in the config file as input_exit_emulator. I could be wrong. It was a little complicated trying to figure out. I've also found that you should remove the keyboard and mouse bindings becomes sometimes they interfere and the hotkeys don't work when you press the buttons, so I've removed all the keyboard and mouse bindings and now there are no issues.

commit ce703955 (2020-09-12): Emuelec-Emulationstation: Bump to 2abc9ab

Fix volume not showing

Fixes #286
closed 2020-09-27 by shantigilbert