EmuELEC/EmuELEC #287: OGA n64 parallel core got black screen

zombiekutez, 2020-09-05:
Device: OGA V1
Emuelec ver. : 3.8-TEST-08292020 64aarch
Core: parallel_n64_libretro (2.0-rc2 3447588368)
I using parallel core and rice pluin, all game after booting go to black screen, sound still running. Mupen64_next is verry better right now but some game like F-zero and racing still need parallel to get it fullspeed.
Here is log:
emuelec logs
> EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 3.8-TEST-08292020 (OdroidGoAdvance.aarch64)
PLATFORM: n64
ROM NAME: /storage/roms/n64/F-ZERO X (U) [!].z64
BASE ROM NAME: F-ZERO X (U) [!].z64
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/n64/F-ZERO X (U) [!].z64
2nd Argument: -Pn64
3rd Argument: --core=parallel_n64
4th Argument: --emulator=libretro
Full arguments: /storage/roms/n64/F-ZERO X (U) [!].z64 -Pn64 --core=parallel_n64 --emulator=libretro --controllers=-p1index 0 -p1guid 19000000010000000100000001010000 
Run Command is:
/usr/bin/retroarch -v -L /tmp/cores/parallel_n64_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/n64/F-ZERO X (U) [!].z64
Emulator Output is:
librga:RGA_GET_VERSION:4.00,4.000000 
ctx=0xdf5e70,ctx->rgaFd=7 
Rga built version:version:+2017-09-28 10:12:42 
rk-debug mali-ver=rsx-p0-1.0
EGL: major=1, minor=4
EGL: Vendor=ARM
EGL: Version=1.4 Bifrost-"r6p0-01rel0"
EGL: ClientAPIs=OpenGL_ES
EGL: Extensions=EGL_KHR_partial_update EGL_KHR_image_pixmap EGL_EXT_image_dma_buf_import EGL_KHR_config_attribs EGL_KHR_image EGL_KHR_image_base EGL_KHR_fence_sync EGL_KHR_wait_sync EGL_KHR_gl_colorspace EGL_KHR_get_all_proc_addresses EGL_IMG_context_priority EGL_ARM_pixmap_multisample_discard EGL_KHR_gl_texture_2D_image EGL_KHR_gl_renderbuffer_image EGL_KHR_create_context EGL_KHR_surfaceless_context EGL_KHR_gl_texture_cubemap_image EGL_EXT_create_context_robustness
EGL: ClientExtensions=EGL_EXT_client_extensions EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_platform_gbm

es logs
> 2017-08-06 16:40:53	DEBUG	fireEvent: game-start /storage/roms/n64/F-ZERO\ X\ \(U\)\ \[\!\].z64 F-ZERO X (U) [!]
2017-08-06 16:40:53	INFO		/emuelec/scripts/emuelecRunEmu.sh /storage/roms/n64/F-ZERO\ X\ \(U\)\ \[\!\].z64 -Pn64 --core=parallel_n64 --emulator=libretro --controllers="-p1index 0 -p1guid 19000000010000000100000001010000 "
2017-08-06 16:41:13	DEBUG	fireEvent: game-end  
2017-08-06 16:41:13	INFO	Window::init
2017-08-06 16:41:13	INFO	Creating window...
2017-08-06 16:41:13	INFO	Created window successfully.
2017-08-06 16:41:13	INFO	Checking available OpenGL extensions...
2017-08-06 16:41:13	INFO	 ARB_texture_non_power_of_two: MISSING
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	INFO	Added known joystick odroidgo2_joypad (instance ID: 1, device index: 0)
2017-08-06 16:41:13	DEBUG	TextureData::load :/scroll_gradient.png
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	DEBUG	ImageIO : rescaling image from 576x324 to 480x270
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/gamepad.svg
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/logos/setup.svg
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/frame.png
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/consoles/setup.png
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/bottom.png
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/bg.png
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/logos/pcengine.svg
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/logos/nes.svg
2017-08-06 16:41:13	DEBUG	ImageIO : rescaling image from 720x720 to 320x320
2017-08-06 16:41:13	DEBUG	ImageIO : rescaling image from 853x480 to 480x270
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/consoles/pcengine.png
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/centerfade.png
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	DEBUG	ImageIO : rescaling image from 320x320 to 160x160
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/shadow.png
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/carousel.png
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/consoles/nes.png
2017-08-06 16:41:13	DEBUG	TextureData::load /storage/.emulationstation/themes/es-theme-EmuELEC-carbon/art/logos/snes.svg
2017-08-06 16:41:13	DEBUG	ImageIO::loadFromMemoryRGBA32
[... 473 lines cut ...]
[INFO] [Environ]: GET_VARIABLE parallel-n64-pak4:
	none
[libretro DEBUG] set filtering mode...
[libretro INFO] mupen64plus: Starting R4300 emulator: Dynamic Recompiler
[libretro INFO] mupen64plus: Init new dynarec
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [config] Saved new config to "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Cheats]: Save game-specific cheatfile: /tmp/database/cht/ParaLLEl N64/F-ZERO X (U) [!].cht
[INFO] [SRAM]: Saving RAM type #0 to "/storage/roms/n64/F-ZERO X (U) [!].srm".
[INFO] [SRAM]: Saved successfully to "/storage/roms/n64/F-ZERO X (U) [!].srm".
[INFO] Content ran for a total of: 00 hours, 00 minutes, 14 seconds.
[INFO] Saving runtime log file: /storage/playlists/logs/ParaLLEl N64/F-ZERO X (U) [!].lrtl
[INFO] [CORE]: Unloading game..
[libretro INFO] mupen64plus: R4300 emulator finished.
[libretro INFO] EmuThread: co_switch main_thread.
[INFO] [CORE]: Unloading core..
[libretro DEBUG] mupen64plus: Stopping emulation.
[INFO] [PERF]: Performance counters (libretro):
[INFO] [CORE]: Unloading core symbols..
[INFO] [Core Options]: Saved core options file to "/storage/.config/retroarch/config/ParaLLEl N64/ParaLLEl N64.opt"
[INFO] Threaded video stats: Frames pushed: 726, Frames dropped: 86.

Thankyou so much for giving :) .

shantigilbert, 2020-09-11:
Parallel-64 does not properly work on Aarch64, but this has been fixed on dev by using the 32bit version

Cee123, 2020-09-11:
I was just about to post the same thing. I am loving the 64-bit version. And so far, that's the only thing I've noticed that doesn't work at the moment - the parallel core. Is there a way to switch to the 32 Bit core?

Cee123, 2020-09-11:
> F-zero and racing still need parallel to get it fullspeed.

F-Zero was one of the handful of games that worked for me at a playable speed using Mupen64plus or Mupen64plus-next. But yeah there are a heap of others that are only playable speed with parallel and the rice renderer.

Cee123, 2020-09-12:
@zombiekutez  Update to the latest test release, it's all good and working now.

shantigilbert, 2020-09-29:
Fixed on v3.8

closed 2020-09-29 by shantigilbert