EmuELEC/EmuELEC #290: WiFi cannot connect to hidden AP but works with visible AP

cederom, 2020-09-06:
Hello world :-) Thank you for this great working out of the box solution for my ODROID GO ADVANCE. I have issues connecting to hidden WiFi AP - it never connects - while connection to visible AP works fine. I guess `wpa_supplicant.conf` could be tuned to use `scan_ssid=1` in order to connect to hidden WiFi networks :-)

JMPMX, 2020-09-09:
Try in the ES menu in the section of networks to turn off and then turn on the network and wait a few seconds, I take it like this, once I have selected it being visible and saved the password, then I have always kept it hidden.

sonninnos, 2021-05-07:
Confirmed. Simply switching SSID broadcasting on in the router made it work immediately. No typos in the manual input.

closed 2025-06-16 by Langerz82