EmuELEC/EmuELEC #293: bluetoothctl is not showing any prompt

aolba, 2020-09-08:
I have some issues trying to pairing a BT controller (steelseries) I try to use bluetoothctl to pairing, but when I use ssh for this, the console freeze, and there is no other prompt.
I use putty and I manage to login to EmuELEC but the command is not working for me.
I use an A95X F3 AIR with S905X3 chip and 2gb ram.
I'm not sure if there is an issue with the BT chip from this box.

aolba, 2020-09-09:
I have emuelec 3.7. The code by itself is not working in my case. After searching, I found a command that help me: systemctl status bluetooth
With this I see bluetooth is off at the moment, maybe that's why my console get freeze. In the same page I see the next command: systemctl start bluetooth, after this I manage to use bluetoothctl, or at least get [bluetooth]# in the console. Now I need to solve other issues with the BT process.

closed 2020-09-09 by aolba

shantigilbert, 2020-09-09:
Why not use the Bluetooth connect in the emulationstation menu?