EmuELEC/EmuELEC #294: emuelec not booting on s905w

elioma91, 2020-09-11:
Emuelec not booting on a Tx3 mini 2/16gb with s905w. The emuelec screen appears, creates partition, restarts automatically. And it stays on that screen forever. Im using the gxl_p281_2g dtb. I tried different dtbs and its the same.

[image]

SamuEDL, 2020-09-11:
Have you tried re-creating the SD with the image?

elioma91, 2020-09-12:
> Have you tried re-creating the SD with the image?

You mean creating an image and write it on the same sd card?

AlienatedDiego, 2020-09-13:
It doesn't seem like a problem with .dtb but with your storage. Format the sd and recreate the image, if it does not work try with a different sd or usb.

elioma91, 2020-09-13:
GOT IT TO WORK FINALLY! Tried different sd cards with different storage capacity and nothing. I tried other dtb that has nothing to do with s905w and it worked *facepalm*

Thanks for the help!!

closed 2020-09-13 by elioma91

elioma91, 2020-09-13:
Solved it. I used another dtb for another processor. Weird how it worked.

shantigilbert, 2020-09-13:
> Solved it. I used another dtb for another processor. Weird how it worked.

Could be that the procesor is really not s905w many Chinese devices do that

kaderr09, 2020-10-05:
> GOT IT TO WORK FINALLY! Tried different sd cards with different storage capacity and nothing. I tried other dtb that has nothing to do with s905w and it worked _facepalm_
> 
> Thanks for the help!!

i have the same problem sir ,could you send me the reference of the dtb that you have tried plz

AlienatedDiego, 2020-10-05:
Check the Amlogic soc on your device to make sure. Without that information it's hard to tell. You can check it from Android using Aida64 app, cpuz or similar. That said, in most cases none of the .dtb for s905w works because it really is a different soc, usually s912.https://github.com/EmuELEC/EmuELEC/wiki/Comprobaci%C3%B3n-Procesador