EmuELEC/EmuELEC #296: How check my compilation (emulator)

zerotrzy, 2020-09-19:
Please help me runing my emulator in qemu or other emulator.
How checking my configuration etc.

shantigilbert, 2020-09-19:
I cannot understand what this has to do with emuelec?

zerotrzy, 2020-09-19:
I have trouble with my config file. I would chceck my image disk before I put my setup to SD card.

Simple, I would run emuelec on desktop linux machine before.

shantigilbert, 2020-09-19:
I honestly have no idea what you are talking about... Sorry, please either explain exactly what yo u are trying to do, if possible with steps. Emuelec does not run on desktop pc

zerotrzy, 2020-09-19:
emuelec can run on desktop if using emulator. for example qemu https://wiki.qemu.org/Documentation/Platforms/ARM

Question is how download kernel, bios etc.

shantigilbert, 2020-09-19:
I have no idea, qemu is not supported

closed 2020-09-19 by shantigilbert