EmuELEC/EmuELEC #297: rt8821cu wlan/bt dongle

cponzoni, 2020-09-20:
usb wlan/bt dongle partially working, wifi working but bluetooth fails to "load firmware".

this dongle is an ezcast 5200, widely sold on amazon, and is based on realtek 8821cu chipset.

`[    4.455924@0] Bluetooth: Core ver 2.22
[    4.456015@0] Bluetooth: HCI device and connection manager initialized
[    4.456027@0] Bluetooth: HCI socket layer initialized
[    4.456033@0] Bluetooth: L2CAP socket layer initialized
[    4.456053@0] Bluetooth: SCO socket layer initialized
[    4.573250@1] Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[    4.573258@1] Bluetooth: hci0: rtl: loading rtl_bt/rtl8821a_config.bin
[    4.573614@1] bluetooth hci0: Direct firmware load for rtl_bt/rtl8821a_config.bin failed with error -2
[    4.573625@1] Bluetooth: hci0: Failed to load rtl_bt/rtl8821a_config.bin
[    4.573627@1] Bluetooth: hci0: rtl: loading rtl_bt/rtl8821a_fw.bin
[    4.713179@0] Bluetooth: hci0: rom_version status=0 version=1
[    4.713206@0] Bluetooth: cfg_sz 0, total size 17428
[    6.798916@1] Bluetooth: hci0 command 0xfc20 tx timeout
[   14.862933@1] Bluetooth: hci0: download fw command failed (-110)
`

**EmuELEC version:**
 - Version 3.8-TEST-09112020

**Device (please complete the following information):**
 - Brand / SOC: ODROID N2+ 4g

shantigilbert, 2020-11-18:
I do not have said HW to properly test, but I will investigate a bit more and see if the FW can be included directly. I assume the files needed are these:

```
[    4.573258@1] Bluetooth: hci0: rtl: loading rtl_bt/rtl8821a_config.bin
[    4.573614@1] bluetooth hci0: Direct firmware load for rtl_bt/rtl8821a_config.bin failed with error -2
[    4.573625@1] Bluetooth: hci0: Failed to load rtl_bt/rtl8821a_config.bin
[    4.573627@1] Bluetooth: hci0: rtl: loading rtl_bt/rtl8821a_fw.bin
```

shantigilbert, 2021-06-07:
Is this still an issue?

lightvil, 2021-06-27:
I tested with OGS and NEXT-531WBT(RT8821CU Dongle) and it worked fine.

shantigilbert, 2021-06-27:
No reply for 21 days I will assume its fixed. Please reopen if this is still an issue

closed 2021-06-27 by shantigilbert