EmuELEC/EmuELEC #298: [Feature Request] 3DO for Android boxes

Cee123, 2020-09-21:
**Is your feature request related to a problem? Please describe.**
Any chance 3DO could be added to amlogic devices like the Android Q Box?

**Describe the solution you'd like**
It works.

**Describe alternatives you've considered**
I've added the core myself, and it works perfect and full speed with the new setting "threaded dsp" turned on.

**Additional context**
I've never been able to play 3DO at full speed on my Android box (except with the real3doplayer emulator)... until now. With this new threaded dsp feature that's been added to the Opera core, I can just turn that on and the good majority of games (about 95%) are running at full speed. No issues at all. I'm amazed at the amount of progress made on this core and the improvements seen in performance during the past year or so.

Cee123, 2020-09-27:
Interestingly, 3DO runs at playable speed on my Android box now even without threaded dsp turned on. It runs at about 57 to 60 fps for most games. Even the tiny frame drops arent that noticeable anymore. Turning on threaded dsp also gives it a big performance boost but because it's experimental can also cause freezing sometimes. I also tried Sega Saturn but my Android box doesn't support Gles3 so it just crashes.

shantigilbert, 2020-09-27:
If its running better now, it could be added yes. what core did you add? how did you made it work? and what version of EmuELEC? (Android is no use for us, so no need to describe the version) or, do you mean it runs ON Android?

Cee123, 2020-09-27:
Oh my God thank you! and my apologies. I left out half the details. Sorry for that. No it wasn't on Android. This was on the latest EmuElec 3.8 aarch64 test image the specific one I used:
EmuELEC-Amlogic.aarch64-3.8-TEST-09122020-Generic

I compiled a 64-bit core using my Odroid N2. 
[opera_libretro.zip](https://github.com/EmuELEC/EmuELEC/files/5287641/opera_libretro.zip)

I placed it into the /core folder on the root. Placed the bios files in the /roms/bios folder. And then manually added options for 3DO in es_systems.cfg . Seems to run fine now, which is weird because it used to run slow.

Cee123, 2020-10-18:
I can maybe upload videos on YouTube and share them here to show that 3DO runs fine on my lower end Android box. There are also settings you can change to make it smoother.

Cee123, 2020-10-19:
Well, has anyone tested it at least?

Cee123, 2020-10-26:
Oh well, I got no answer here. So I might take some videos of it working as soon as I can get a decent camera. And then put them on YouTube along with the instructions on how to get this and Sega Saturn working on their TV box running EmuElec, as some people ARE interested in these two systems and are curious to see how they run on their devices.

shantigilbert, 2020-10-26:
Or maybe nobody had time to test it? not many people that help the project uses a S905 anymore. PRs are always welcome.

Cee123, 2020-10-26:
> Or maybe nobody had time to test it? not many people that help the project uses a S905 anymore. PRs are always welcome.

Oh ok then. Im sorry. You could've told me. I didn't realize they weren't all that common anymore.

shantigilbert, 2020-10-26:
Oh they are common as dirt, but read the part where I said that people that **help out** probably do not have one, not many people help out by testing, reporting, coding, etc

Cee123, 2020-10-26:
> Oh they are common as dirt, but read the part where I said that people that **help out** probably do not have one, not many people help out by testing, reporting, coding, etc

Oh ok sorry i misunderstood. Yeah i get what you mean now, that actually makes sense then. No worries.

dm2912, 2020-10-26:
If you post FULL tutorial on how to do it, i will try on mine, but needs full step by step instructions for dummies :D

shantigilbert, 2020-10-28:
@Cee123 I tried 3do on the s905, not sure why you call this "playable" not even gex could be played at full speed. Can you post your settings?

Cee123, 2020-10-28:
> 
> 
> @Cee123 I tried 3do on the s905, not sure why you call this "playable" not even gex could be played at full speed (45fps and its slow). Can you post your settings?, I just used default

Ok yeah, it's probably not full speed on default settings or even close to it. I guess it depends on the specs. But you can get a get a performance boost by enable the setting "threaded dsp" under Quick Settings > Options. And to me that feels like full speed or close to it for a lot of games. It depends on the game. But only problem with that setting is that it isn't 100% stable 100% of the time. So I'm not sure but I think the feature is a work in progress.

shantigilbert, 2020-10-28:
The only 2 games I have crashed with "threaded dsp" on. With that said, if I add it  it will be unsupported in all aspects, the main reason why I don't want to include such low performance core in these devices is mostly maintenance, people will complain that it does not run at 60 fps. Anyways I will probably enable it for v4.0 but I am still undecided.

Cee123, 2020-10-29:
> The only 2 games I have crashed with "threaded dsp" on. With that said, if I add it it will be unsupported in all aspects, the main reason why I don't want to include such low performance core in [...]

Oh ok, no worries then. Yeah you're right. No point adding it then for the time being if it's not up to scratch.

Cee123, 2020-10-29:
@dm2912 So your Android box is the same as mine I'm guessing. Have you noticed any graphical glitches in Flycast and black textures in Mupen64plus/NX? Another user AlienatedDiego said he also noticed the same thing.

https://github.com/EmuELEC/EmuELEC/issues/299

dm2912, 2020-10-29:
> @dm2912 So your Android box is the same as mine I'm guessing. Have you noticed any graphical glitches in Flycast and black textures in Mupen64plus/NX? Another user AlienatedDiego said he also noticed the same thing.
> 
> #299

My box is a Tanix TX3 4gb ram, S905X3

as far as dreamcast goes, only things I play on DC are Crazy Taxi 1 & 2, Soul Calibur, Skies of Arcade and Jet Grind Radio

Crazy Taxi 1 is fine, fast and smooth
Crazy Taxi 2 seems slow, but that might be intended... the graphics are noticeably better
Soul Calibur runs great
Jet Set Radio runs fine
Skies of Arcadia, normal gameplay is fine but it stutters a bit on the end of battle video and other cut scenes from time to time

But to be honest Dreamcast is probably pushing the limits of these boxes so im pretty happy with it. 

Should also note that my box is running at 40c when playing DC games so there's no thermal issues for me, performance might be different if cpu is getting hot

Cee123, 2020-10-29:
Oh ok, so yours is a different model. You dont get any glitches in Soul Calibur? That's pretty cool that they all work great on yours. I get very good performance on mine they all run great as well, no issues whatsoever. Except the graphics glitching on some of them. So you probably dont have most the games ive mentioned on that thread (also those Naomi games). That's alright. Im not worried about performance, that's all fine. Just the glitching.

shantigilbert, 2020-10-29:
Can you guys keep it in topic, open another issue for Flycast  please. 

Opera will be included in v4.0 with a big warning, if people want to play this in slow motion thats on them :)

Cee123, 2020-10-30:
It doesn't seem to run that badly on mine. It's definitely a lot better than on my Raspberry Pi 3B. And a number of games don't crash with that setting turned on. I have about 45 games that I've tested overall. I really need to make a couple of videos to show it. :(

No worries, I couldn't tag him in the other thread. 👍

Cee123, 2020-11-10:
As promised here's a couple of videos I took of a few games running on my Android Q Box. Videos I took with my new camera. Apologies in advance for the sound quality. Still have to figure out settings on this thing.

Crash N Burn (Freezes with Threaded Rendering)
https://www.youtube.com/watch?v=rPDn91mOAAg

Captain Quazar
https://www.youtube.com/watch?v=uiqPm2e1aDI

Corpse Killer
https://www.youtube.com/watch?v=2YPApJEikoo

Gex
https://www.youtube.com/watch?v=_e2kQS2ykb4

Mega Race
https://www.youtube.com/watch?v=sELDhvLRQT0

PO'ed
https://www.youtube.com/watch?v=XSj2V9v9Z48

Road Rash
https://www.youtube.com/watch?v=D5BktNw6T5w

Cee123, 2020-11-17:
> Opera will be included in v4.0 with a big warning, if people want to play this in slow motion thats on them :)

Did you see the videos I uploaded?

Cee123, 2020-11-18:
@shantigilbert I've just provided you with proof of several games running at a playable speed and how to achieve that result and you didnt even bother to check them out.

Cee123, 2020-11-18:
But, i will admit, there are also some games ive tried that are painfully slow. For example FIFA Soccer and Need for Speed. But anyways thanks for adding the core @shantigilbert .

closed 2020-11-18 by Cee123

shantigilbert, 2020-11-18:
Please dome be that guy! I added the core because you requested, does not matter if I saw the videos or not (which I did) I don't spent 24/7 working on emuelec, I make no money on it, need my real job and my personal time.

Cee123, 2020-11-18:
> Please don't be that guy! I added the core because you requested, does not matter if I saw the videos or not (which I did) I don't spent 24/7 working on emuelec, I make no money on it, need my real job and my personal time.

I just asked you a simple question. What's so difficult? You've been replying to everyone else here, but not me. And I've been nice to you, ive been patient with you, ive provided details that youve asked for, and you havent even bothered to check my responses at all. And at one point even telling me that no one on the Emuelec team likely has an amlogic device like mine.

shantigilbert, 2020-11-18:
Point is, I don't own you nothing, nor work for you.

commit 5c14fb41 (2020-11-02): Opera (3DO): Icnlude in all projects

although 3DO does not run that good on s905 devices, some people want to have it enabled. Fixes #298
commit 40859f6b (2020-11-02): Opera (3DO): include in all projects

although 3DO does not run that good on s905 devices, some people want to have it enabled. Fixes #298