EmuELEC/EmuELEC #299: Graphics glitches seen in Flycast and Mupen64plus/Next on Android Q Box

Cee123, 2020-09-21:
**Describe the bug**
On Flycast, both Dreamcast and Naomi games can be glitchy. Most games are fine, or have a tiny glitch here and there. But some games are so bad they are basically unplayable. Some examples of what I mean by glitches are like missing textures, flickering, flashing and garbled graphics. Games for example Dynamite Cop, House of the Dead 2, Sega Bass Fishing, Unreal Tournament and Naomi games such as Asian Dynamite, Heavy Metal Geomatrix, both Melty Blood games are unplayable due to massive glitching.

And on Mupen64plus and Mupen64plus Next, there is an issue (not in all games but quite a lot) where distant textures are blacked out or darkened until you get right up close to them. But they are still dark. An example where you would get blackened textures would be in Banjoe Kazooie, Banjo Tooie, Donkey Kong 64, San Francisco Rush.

I'll try and get some photos of these and share them here.

I'm using the arch64 test build.

**To Reproduce**
Android Q Box
OpenGles2 I think is the only supported option
Previous builds of Emuelec, up until the latest Arch64 test build
Seen in cores
- Flycast
- Mupen64plus
- Mupen64plus-next

**Expected behavior**
Should run fine.

**Screenshots**
In the process of taking some photos, I'll add them up there.

**EmuELEC version:**
All the versions I've tried.

**Device (please complete the following information):**
Amlogic
Q Box/Q Box Clone
Android 5.1.1

**Additional context**
Not sure. Any questions I'll be happy to help.

shantigilbert, 2020-09-22:
what CPU is this? I do not know anything about "Q-box"

Cee123, 2020-09-22:
> 
> 
> what CPU is this? I do not know anything about "Q-box"

Ah yeah, of course sorry. I just looked up the specs for the Android box. I've updated the OP with the details too.

Android 5.1
Amlogic s905 quad-core cortex-A53 2.0 Ghz
Mali-450 5-Core GPU
2GB DDR3 16GB EMMC
2.4G +5G WIFI,802.11 B/G/N
100/1000M LAN
Bluetooth4.0
Hardward 3D graphics acceleration
2K*4K Support
Dolby/ 3G/ SPDIF/ IEC958/ Coaxial

Cee123, 2020-09-24:
This is the device I have. Well actually it's a clone of that with exactly the same specs and everything.
https://www.geekbuying.com/item/Q-BOX-Amlogic-S905-4K-KODI-TV-Box-2G-16G-2-4G-5G-WIFI-1000M-LAN-Bluetooth4-0-KODI-Dolby-364735.html

Cee123, 2020-09-24:
I also have one of these, the Q Plus Android box
https://www.amazon.com.au/Android-Quad-core-Support-Decoding-EASYTONE/dp/B07QKJPPTF

Emulators would run much better on that, as it's better hardware but EmuELEC is not compatible with it.

Cee123, 2020-09-28:
I think these few issues could be related to libhybris, like it is on Odroid N2.
> Amlogic-ng: Goodbye libhybris! - Use proper GLES headers, this fixes some gfx errors on some cores, like Mupen64

shantigilbert, 2020-09-28:
No, this is different, I just haven't had time to test my s905, since its in a box somewhere.

shantigilbert, 2020-10-01:
didn't you said you could not run Mupen64plus-next?

shantigilbert, 2020-10-01:
I just tested all of this on my s905 and I do not have any glitches on the games I tried, if this is only on certain games, this might be an emulator issue and not something I can fix on EmuELEC. Granted I do not have many games as you to test. 

Mupen64plus-next is already fixed, at leas the black screen bug you reported on #300 

You could also test v3.8 (not aarch64) and see if you have the same issues there.

Cee123, 2020-10-01:
> 
> 
> didn't you said you could not run Mupen64plus-next?

Yeah well I mean in the previous versions when it worked. Sorry.

Cee123, 2020-10-01:
> 
> 
> I just tested all of this on my s905 and I do not have any glitches on the games I tried, if this is only on certain games, this might be an emulator issue and not something I can fix on EmuELEC. Granted I do not have many games as you to test.
> 
> Mupen64plus-next is already fixed, at leas the black screen bug you reported on #300
> 
> You could also test v3.8 (not aarch64) and see if you have the same issues there so we can narrow it down.

Really? That's interesting. Yeah it seems to be only on certain games, whereas the rest of them run perfectly fine. You could be right, it could be down to some issue in the emulators themselves, or even down to certain settings. But I have left them at default. I've tried setting different settings to see if it makes a difference. Interestingly, the black textures thing seen in Mupen doesn't exist with Parallel. So I'd just use Parallel instead for those games. But if it doesn't do that on yours then it must be mean it's something else, or even to do with the emulator/s. Could post it as a bug on Mupen and Flycast libretros.

Cee123, 2020-10-19:
I don't know there's anything that can be done to fix this. Did you test out the games that I mentioned, or other ones? Because it seems to be only in those specific games that I mentioned. I'm now thinking it either could be certain drivers or have something to do with this Android Box's inability to render OpenGles 3.0 which perhaps has elements used by certain games on Flycast or Mupen.

AlienatedDiego, 2020-10-19:
I have an s905x and that problem, as you say, is reproduced in some flycast games on old generation Amlogic s905 devices with mali 450. It seems that the drivers are not compatible with the current code. maybe the developer can fix that, but ... I don't think so. Dreamcast emulation seems a bit stagnant now.

Cee123, 2020-10-19:
> I have an s905x and that problem, as you say, is reproduced in some flycast games on old generation Amlogic s905 devices with mali 450. It seems that the drivers are not compatible with the current [...]

Ahhh ok, interesting. Yeah i didnt think the issue would be confined to one particular tv box. Glad its not just mine and that you've also confirmed this. So it's a driver thing? I thought it could've been something to do with that too. Yeah seems to be a bit hit or miss with Flycast. But fortunately most games don't have these glitches. Have you also noticed a blackening of textures in mupen64plus and mupen64plus-next? Parallel seems to be glitch free.

AlienatedDiego, 2020-10-19:
I do not know it, I use Parallel at 640x480 almost always and I do not see glitches problems except if I have activated the rice plugin, also now in s905x3.

Cee123, 2020-10-19:
> I do not know it, I use Parallel at 640x480 almost always and I do not see glitches problems except if I have activated the rice plugin, also now in s905x3.

Oh ok, nice. Yeah ive been using parallel as well for most games. Some games like Banjo kazooie and banjo tooie ran too slow even with parallel so i had to turn on rice for those.

shantigilbert, 2020-11-26:
Is this still an issue? I  still cannot reproduce it on my boxes, maybe try the newest versions?

Cee123, 2020-11-27:
It still does it. I'll try the newest versions and see what happens.

Cee123, 2020-11-27:
Just tried the test version 4 and tested out a particular game that's unplayable due to this issue - Cannon Spike for Naomi, seems to be the same. This is one of the small handful of games with the issue.

[image]

[image]

I also tried Mupen64plus and Mupen NX to see if they had that blackened textures look, seems to be the same also. It's not such a big deal, since I can fix the problem just by using Parallel instead.

[image]

[image]

[image]

[image]

It could be an issue with drivers, or even with this box's ability to accurately use Open GLES.

AlienatedDiego, 2020-11-27:
Still there .in my case using s905x / mali 450 graphics .to reproduce in flycast running some like Soul Calibur

Cee123, 2020-11-27:
> 
> 
> Still there .in my case using s905x / mali 450 graphics .to reproduce in flycast running some like Soul Calibur

Yeah Soul Calibur is another one. Seems to be a bit of glitching there.

shantigilbert, 2020-11-27:
I think this is related to emulators and not emuelec, Flycast "might" gtet fixed in the future: https://github.com/libretro/flycast/issues/985#issuecomment-731055385 but for Muppen I have no idea.

AlienatedDiego, 2020-11-27:
In my tests (flycast), that problem in some games was always there, at least since my inception in 3.1. I have always believed that it is something related to emulator / mali 450 graphics. This does not happen on NG devices.

Apart from this, EmuELEC has an outstanding general performance in Amlogic Project, taking into account that in s905x it barely ran android 6.0 well

For my part the thread can be closed; maybe if there are news on flycast it could reopen.

Cee123, 2020-11-27:
Interesting. I just tested out Flycast on retroarch for Android and same thing happens. So it's not just EmuElec. But I couldn't test mupen nx because i couldn't get it to stop crashing.

kage2051, 2020-12-30:
> Interesting. I just tested out Flycast on retroarch for Android and same thing happens. So it's not just EmuElec. But I couldn't test mupen nx because i couldn't get it to stop crashing.

The libretro core has been updated with a fix for Mali-400/450. Worth testing the new build.

Cee123, 2020-12-30:
> 
> 
> > Interesting. I just tested out Flycast on retroarch for Android and same thing happens. So it's not just EmuElec. But I couldn't test mupen nx because i couldn't get it to stop crashing.
> 
> The libretro core has been updated with a fix for Mali-400/450. Worth testing the new build.

Really? Wow, that's great! Fantastic! But which core? Flycast or Mupen Next?

Cee123, 2020-12-30:
> > Interesting. I just tested out Flycast on retroarch for Android and same thing happens. So it's not just EmuElec. But I couldn't test mupen nx because i couldn't get it to stop crashing.
> 
> The libretro core has been updated with a fix for Mali-400/450. Worth testing the new build.

Oh i see what you mean now. On the latest EmuElec test build, using the older version of flycast. Cool. I might test it out.

Cee123, 2021-01-11:
I tried the next test release but it's still the same. Glitches are still there.

Langerz82, 2023-02-08:
Is this still an issue with 4.6?

multigamesystem, 2023-07-15:
Any word on how to fix this on AML-S905X-CC (Le Potato)?  Dreamcast games run really well aside from these weird sky box or background glitches.

Cee123, 2024-03-01:
I fixed the shadow/dark patches problem on mupen64plus-next GLES2 by disabling the setting for LOD Emulation under the core options.