EmuELEC/EmuELEC #300: Mupen64plus-Next black screen on latest Emuelec test build - Android Q Box

Cee123, 2020-09-21:
**Describe the bug**
Just letting you guys know, Mupen64plus-next loads to a black screen on the Android Q Box. I have a theory. I think this brand of Android Box is only compatible with OpenGles2 whereas the core version is build for OpenGles3.

I'm using the latest Arch64 test build. The "generic" version.

**To Reproduce**
-

**Expected behavior**
Should load, but doesn't.

**Screenshots**
-

**EmuELEC version:**
 3.8-TEST-09112020

**Device (please complete the following information):**
Amlogic
Q Box/Q Box Clone
Android 5.1.1

**Additional context**
Anyway I can compile the OpenGles2 version of Mupen Next? Or someone can provide it? I think the previous version of Emuelec that I tried had the OpenGles2 version and it worked fine.

Cee123, 2020-09-22:
Well I just compiled a few Mupen64plus-next cores on Ubuntu using my Odroid N2 so I could get them arch64. And I've tried both cores for opengles2 and opengles3 and using LLE and without. And each time I try to get these cores to work, it appears to work for a second or so but then it crash dumps resulting in a Segmentation Fault.

[image]

[mupen64plus_next_libretro_gles2(LLE).zip](https://github.com/EmuELEC/EmuELEC/files/5259127/mupen64plus_next_libretro_gles2.LLE.zip)
[mupen64plus_next_libretro_gles2.zip](https://github.com/EmuELEC/EmuELEC/files/5259130/mupen64plus_next_libretro_gles2.zip)
[mupen64plus_next_libretro_gles3(LLE).zip](https://github.com/EmuELEC/EmuELEC/files/5259131/mupen64plus_next_libretro_gles3.LLE.zip)
[mupen64plus_next_libretro_gles3.zip](https://github.com/EmuELEC/EmuELEC/files/5259133/mupen64plus_next_libretro_gles3.zip)

Not sure what I could do to get this core to work.

asakous, 2020-09-22:
use old version of  mupen64plus-next . probably before this commit tag (9f3169).

Cee123, 2020-09-22:
> 
> 
> use old version of mupen64plus-next . probably before this commit tag (9f3169).

Awesome, thank you! Alright I'll try one of the older versions before that commit tag then, I'll compile one or two of those and I'll see if they work.

Cee123, 2020-09-22:
> 
> 
> use old version of mupen64plus-next . probably before this commit tag (9f3169).

I couldn't get an older version to build for some strange reason. I'll keep trying.

asakous, 2020-09-22:
> > use old version of mupen64plus-next . probably before this commit tag (9f3169).
> 
> I couldn't get an older version to build for some strange reason. I'll keep trying.

try remove PKG_SHA256 in package.mk. maybe works

Cee123, 2020-09-22:
> 
> 
> > > use old version of mupen64plus-next . probably before this commit tag (9f3169).
> > 
> > 
> > I couldn't get an older version to build for some strange reason. I'll keep trying.
> 
> try remove PKG_SHA256 in package.mk. maybe works

Ok, thank you. I'll try that and see if it works.

Cee123, 2020-09-22:
I tried an older version, the one just before 9f3169 and I tried changing the flags in the Makefile to set them to Gles2 and Gles3. I tried a few different versions. But I keep getting an error now that says error missing libGL.so.1

asakous, 2020-09-23:
please reference my package settings [https://github.com/asakous/Neo-EmuELEC-H3/tree/master/packages/sx05re/emulators/mupen64plus-nx](url)

Cee123, 2020-09-24:
> 
> 
> please reference my package settings [https://github.com/asakous/Neo-EmuELEC-H3/tree/master/packages/sx05re/emulators/mupen64plus-nx](url)

I'm not sure I understand. Do I post the issue there and reference it from here? I click on the link but nothing comes up.

asakous, 2020-09-24:
my bad . typo.  
you can copy the url and paste to chrome.
and use my package.mk and patches  to build mupen64plus-nx.

Cee123, 2020-09-24:
> 
> 
> my bad . typo.
> you can copy the url and paste to chrome.
> and use my package.mk and patches to build mupen64plus-nx.

That works now. Sorry about that. Thanks so much!! I'll try that and see if I can get it to work. Much appreciated!

Cee123, 2020-09-24:
I'm really sorry for being a noob when it comes to github and probably Linux as well. But I'm not sure exactly what I'm supposed to do. So basically I've downloaded these files and placed them in a folder mupen64plus-next, so do I just build it from the package.mk file? And if so, what command do I use? I've tried make but it's not working. Or do I fetch EmuELEC from the github, fetch mupen64plus-next and add it to that folder and build it from there? Or do I fetch mupen64plus-next from the libretro github and add it to that folder and build it from there? Sorry I'm still trying to figure this out.

asakous, 2020-09-24:
you need to fetch EmuELEC from the github and replace mupen64plus-next with my files. and build the whole system.
or simply go to [https://github.com/asakous/Neo-EmuELEC-H3/releases/tag/V1.1](https://github.com/asakous/Neo-EmuELEC-H3/releases/tag/V1.1)  and download mupen64plus_next_libretro.zip . hopefully it works on Amlogic  system.
otherwise you just wait to shantigilbert to fix it.
 .

Cee123, 2020-09-24:
> you need to fetch EmuELEC from the github and replace mupen64plus-next with my files. and build the whole system.
> or simply go to https://github.com/asakous/Neo-EmuELEC-H3/releases/tag/V1.1 and [...]

Oh my God thank you so much! You're awesome. I just downloaded the zip file it was a lot easier for me. Thank you.

Cee123, 2020-09-24:
Interesting, seems now there's a different issue. An error message saying "wrong ELF class".

[image]

asakous, 2020-09-25:
that means the binary file is not compatible with your system. so I am out of ideas . you better wait for shantigilbert to fix it.

Cee123, 2020-09-26:
> 
> 
> that means the binary file is not compatible with your system. so I am out of ideas . you better wait for shantigilbert to fix it.

Thanks for all your help. Yeah I think I'll just wait for a fix hopefully in the next version or whatever.

Cee123, 2020-09-28:
I think that core could be 32-bit. Because it says wrong ELF class: ELFCLASS32.

Cee123, 2020-09-29:
So no one knows what's causing the segmentation fault on the mupen64plus-next cores i compiled? Possibly a driver issue?

shantigilbert, 2020-09-29:
> 
> 
> So no one knows what's causing the segmentation fault on the mupen64plus-next cores i compiled? Possibly a driver issue?

if you are compiling them I think its better to ask in the mupen64 repo.

shantigilbert, 2020-10-01:
Try this one please

[mupen64plus_next_libretro.zip](https://github.com/EmuELEC/EmuELEC/files/5309014/mupen64plus_next_libretro.zip)

Cee123, 2020-10-01:
> 
> 
> Try this one please
> 
> [mupen64plus_next_libretro.zip](https://github.com/EmuELEC/EmuELEC/files/5309014/mupen64plus_next_libretro.zip)

Awesome. I will try that version and see if it works. Thanks!

Cee123, 2020-10-01:
> 
> 
> Try this one please
> 
> [mupen64plus_next_libretro.zip](https://github.com/EmuELEC/EmuELEC/files/5309014/mupen64plus_next_libretro.zip)

Thank you! You're a legend. Can confirm that version 100% works. Thanks for that!

shantigilbert, 2020-10-29:
This has been fixed on v3.9

closed 2020-10-29 by shantigilbert