EmuELEC/EmuELEC #301: librga missing make image failed

LightenPan, 2020-09-26:
**Describe the bug**
hi shantigilbert:

i run `PROJECT=Rockchip DEVICE=OdroidGoAdvance ARCH=arm make image`

it runs errors because the rockchip librga is missing, The error message is as follows.

can you fix it?

--2020-09-26 12:38:01--  https://github.com/rockchip-linux/linux-rga/archive/72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Resolving github.com (github.com)... 13.250.177.223
Connecting to github.com (github.com)|13.250.177.223|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-09-26 12:38:02 ERROR 404: Not Found.

--2020-09-26 12:38:02--  http://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz [following]
--2020-09-26 12:38:03--  https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-09-26 12:38:04 ERROR 404: Not Found.

--2020-09-26 12:38:04--  https://github.com/rockchip-linux/linux-rga/archive/72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Resolving github.com (github.com)... 13.250.177.223
Connecting to github.com (github.com)|13.250.177.223|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-09-26 12:38:05 ERROR 404: Not Found.

--2020-09-26 12:38:05--  http://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz [following]
--2020-09-26 12:38:05--  https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-09-26 12:38:06 ERROR 404: Not Found.

--2020-09-26 12:38:06--  https://github.com/rockchip-linux/linux-rga/archive/72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Resolving github.com (github.com)... 13.250.177.223
Connecting to github.com (github.com)|13.250.177.223|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-09-26 12:38:07 ERROR 404: Not Found.

--2020-09-26 12:38:07--  http://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz [following]
--2020-09-26 12:38:07--  https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-09-26 12:38:08 ERROR 404: Not Found.

--2020-09-26 12:38:08--  https://github.com/rockchip-linux/linux-rga/archive/72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Resolving github.com (github.com)... 13.250.177.223
Connecting to github.com (github.com)|13.250.177.223|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-09-26 12:38:09 ERROR 404: Not Found.

--2020-09-26 12:38:09--  http://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz [following]
--2020-09-26 12:38:09--  https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443... connected.
[... 90 lines cut ...]
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz [following]
--2020-09-26 12:38:22--  https://sources.libreelec.tv/mirror/librga/librga-72e7764a9fe358e6ad50eb1b21176cc95802c7fb.tar.gz
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-09-26 12:38:23 ERROR 404: Not Found.

**To Reproduce**
PROJECT=Rockchip DEVICE=OdroidGoAdvance ARCH=arm make image

**Expected behavior**
make image successful

**Screenshots**
no need

**EmuELEC version:**
 - all Version

**Device (please complete the following information):**
OdroidGoAdvance 

**Additional context**

shantigilbert, 2020-09-26:
Use this repo: https://github.com/Caesar-github/linux-rga

LightenPan, 2020-09-26:
think you, it works.

commit d68ba0c9 (2020-09-27): librga: bump to 261c7ba

and change repo. Fixes #301
closed 2020-09-27 by shantigilbert