EmuELEC/EmuELEC #302: Disk available space in /storage partition is wrong

ws02589111, 2020-09-27:
**Describe the bug**
The available space is different between get by df -h and get in es system info->disk usage.

**To Reproduce**
Steps to reproduce the behavior:
1. ssh to box, run df -h, get the /storage partition space info.
2. es, press start， system info->disk usage.

**Expected behavior**
The available space should equal

**Screenshots**
[image]
[image]

**EmuELEC version:**
 - Version [e.g. 22]
latest dev

**Device:**
 - SOC:  S905M

**Additional context**
I found my /storage partition missing about 5% disk space， the Available + Used not equal to Size， is this the problem of ext filesystem？Maybe the bug is cause by it

shantigilbert, 2020-09-28:
not really an issue, since both numbers are estimates (rounded), to get the real values you would need to calculate the bytes directly. 

No idea if its a bug or not, in my case the difference is 1% but the numbers are the same 

system info: 
57.10 total
55.1 used 
96% 

df -h 
`/dev/mmcblk0p2           57.1G     55.1G      1.9G  97% /storage`

ws02589111, 2020-09-28:
thx your reply.
I have found the issue why missing 5%  [ext filesystem reserved blocks](https://wiki.archlinux.org/index.php/Ext4#Remove_reserved_blocks).

I try  to set reserved block to 1%
[image]
[image]
now ，my space is back。

so i suggest to add it to init script after resize

TheLastProject, 2020-09-28:
I would very strongly urge against setting reserved blocks to 0%. I don't know from the top of my head as which user EmuELEC runs but the OS definitely does writes to /storage and depending on the user it runs at you may very well run into issues writing savefiles or other things. You will also lose performance due to fragmentation if the disk gets too full. Lots of painful hard-to-debug issues.

See also https://wiki.archlinux.org/index.php/Ext4#Reserved_blocks.

>It is generally safe to reduce the percentage of reserved blocks to free up disk space when the partition is either:
> - Very large (for example > 50G)
> - Used as long-term archive, i.e., where files will not be deleted and created very often

Your storage (and that of most users) is neither of those.

ws02589111, 2020-09-28:
Ok, i try to set my 32g card to 3% and 64g card to 2% to test several days。
if the card size > 64gb,  the 5% of the available  space will become a lot

shantigilbert, 2020-09-29:
Yeah, no,  this is not going to happen, its a very bad idea and most likely will end up in corrupted installations.

closed 2020-09-29 by shantigilbert