EmuELEC/EmuELEC #305: is it possible to add Gpio to your neo emuelec system?

ctogamez, 2020-09-30:
Hello, asakous, all good.

My name is Fabio, and as a player, I use the nickname cto gamez. I am Brazilian and I love oldgen games.

I met this week, your neo emuelec project, and I was very surprised by this system, because it is doing something that I have not seen in other systems, running naomi system very fluidly in many games.
But with USB controls there is always a delay when playing. And I would like to know if it is possible to implement GPIO in your system, as I guarantee that it would be very good.

I thank you for the effort you have made in the scene with orange pi.
and my congratulations, and continue on this beautiful project

Ass: Cto Gamez

TheLastProject, 2020-09-30:
Duplicate of #304 and again wrong repo, this is EmuELEC not Neo EmuELEC. You are probably looking for https://github.com/asakous/Neo-EmuELEC-H3.

ctogamez, 2020-09-30:
thanks (y)

closed 2020-09-30 by shantigilbert