EmuELEC/EmuELEC #306: OGA Background image scaling looks pixelated in 3.7 and 3.8, was fine in 3.6

mbroers, 2020-09-30:
On Odroid Go Advance, the background images are not scaling properly in the latest two releases 3.7 and 3.8.  In 3.6 the scaling looks fine for the same image file.

3.8 with background pixellation:
[image]

3.6 looking fine:
[image]

I've tried disabling the theme option 'OPTIMIZE FOR SMALL SCREENS' and it doesnt resolve the issue.

shantigilbert, 2021-02-14:
Is this still an issue on v4.0?

mbroers, 2021-02-17:
sadly yeah.. i havent verified on n2+ yet

mbroers, 2021-02-17:
when i switch scaling to 2x1 or 3x1 the images look fine, but 1x1 they scale like this

shantigilbert, 2021-02-17:
Ok, then sorry I really do not know why this is happening or how to fix it.

mbroers, 2021-02-17:
It doesnt happen in crystal but I am unable to find a UI style in crystal that duplicates that old oga default style that worked for me in 3.6. 1x1 in crystal has a bunch of text and doesnt pull the marquee.  Maybe I will petition dim and describe what Im seeking or help collab on that project if possible.

mbroers, 2021-02-17:
I think I'm losing my mind.  es-theme-emuelec-carbon is fine now. no more pixellation on scaling.  i swear i had this on different themes after recently setting up 4.0 in 1x1 configs but this one is fine now.  close it!

dm2912, 2021-02-17:
> It doesnt happen in crystal but I am unable to find a UI style in crystal that duplicates that old oga default style that worked for me in 3.6. 1x1 in crystal has a bunch of text and doesnt pull the [...]

is small panel 3:2 no good for the oga?

[image]

that was specifically done to mimic a single carousel, but using a single image rather than loading multiple elements

And for gamelist similar.. 

[image]

just hit me up on discord in the #theming bit if there's a particular layout you want

shantigilbert, 2021-02-17:
Also Crystal should be default on v4.0 and it also scales accordingly to the device used, so small panel 3:2 for OGA, small panel 16:9 for OGS

closed 2021-02-17 by shantigilbert

mbroers, 2021-02-17:
I will check your suggestions, thanks! My personal preference was es-theme-emuelec-carbon specifically for the default 1x1 browsing with the marquee text over box art that fades to video with the marquee moving up and smaller.. I really thought it was perfect but I dig the theme work for crystal very very much. Would be awesome to have such a style in crystal too!  I'll play around but the reason I opened this issue is gone as far as I can tell.

dm2912, 2021-02-17:
> I will check your suggestions, thanks! My personal preference was es-theme-emuelec-carbon specifically for the default 1x1 browsing with the marquee text over box art that fades to video with the [...]

thats basically what small panel does, except it uses screenshot with marquee on top as opposed to box art. 

honestly, the reason i went screenshot route is the new icons for favourites and cheevo's are really hard to line up with boxart which can be all sorts of random shapes. 

without getting too sidetracked, if you go to line 908 of gamelist.xml in crystal you can swap out >image< for >thumb< and get a similar effect..

but no need now carbon is working :D