EmuELEC/EmuELEC #310: Flycast bug with button on Emuelec

hehk1234, 2020-10-02:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version [e.g. 22]

**Device (please complete the following information):**
 - SOC: [e.g. S095x3]
 - Brand: [e.g. Beelink]
 - Model [e.g. mini g1]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

shantigilbert, 2020-10-02:
Disable analog to digital on Retroarch. Please only open BUG issues here, this is a user error.

closed 2020-10-02 by shantigilbert

hehk1234, 2020-10-02:
Sir, the problem is for odroid go advance, retroarch don't give me option
to change "analog - digital" I already try. Another problem, reicast don't
run any rom here, only flycast.

shantigilbert, 2020-10-02:
Rgb10 is not supported, only odroid go advance.

hehk1234, 2020-10-02:
but they have same hardware, RGB10 is clone from odroid.... all emulators
works fine, I just get the problem with gamepad on dreamcast (and plugin
too). Sorry I don't want disturb you, I try everything to get gamepad set
config, but fails.... it recognize analog like D-pad.

shantigilbert, 2020-10-02:
Go ask the manufacturer then, only odroid go advance is supported.

hehk1234, 2020-10-02:
Ok bro, I test o batocera and works (with same IMG for odroid advance) . I
like more emuelec because it have more compatibility on rgb10 and shaders
filters.