EmuELEC/EmuELEC #312: Error found in version 3.8 (s905w and s905x3)

rochajunior, 2020-10-02:
**Describe the bug**

I don't know if I'm in the right place to report an error found in version 3.8 (s905w and s905x3). In the ES menu, the system slows down and when entering All Games, it simply locks the system.

**EmuELEC version:**
 - Version [3.8]

**Device (please complete the following information):**
 - SOC: [S905W e S095x3]
 - Brand: [Tanix]
 - Model [Tx9, tx3 e X96]

**Additional context**

I perceive an incredible effort on the part of the developer to make the systems more compatible and smoother in running the games, the best version of emuelec until today was 3.5. Version 3.8 is promising, but the errors of slowness and crashes when entering and exiting games need to be corrected. thanks to everyone involved for this wonderful system.

AlienatedDiego, 2020-10-02:
You may be trying to handle thousands of roms (all systems) with netplay enabled overloading the system. If so, try disabling it in the game options, it is also important to use a fast sd memory.

rochajunior, 2020-10-02:
> You may be trying to handle thousands of roms (all systems) with netplay enabled overloading the system. If so, try disabling it in the game options, it is also important to use a fast sd memory.

Yes, the netplay function is active ... but the only roms are the ones shown in the image, the ports!

AlienatedDiego, 2020-10-02:
It is understood. In that case I don't know what the solution is. But in my tests test 3.8 at least for s905x3 I have not seen problems. The system is fluid. Make sure your 3.8 image is official from github. the issue will be open until reviewed by the developer.

dm2912, 2020-10-03:
Just an observation. On the TX3 are you using a mechanical hard drive for roms?

I have the TX3 and have discovered it constantly sends hard drives to sleep. I was having all sorts of freezing issues, even with system on SD and roms on mechanical hard drive it was still freezing. As soon as i moved things to flash based storage (so no sleep) its been running fine. 

Its an amazing box, but is very sleepy :D

AlienatedDiego, 2020-10-03:
> Just an observation. On the TX3 are you using a mechanical hard drive for roms?
>
> I have the TX3 and have discovered it constantly sends hard drives to sleep. I was having all sorts of freezing [...]

Same for me (tx3 s905x3 with mechanical hdd), defragmenting disk and using usb 3.0 box helped, only in a few dreamcast games it still happens sometimes.

rochajunior, 2020-10-03:
I am not using a hard disk (HD) I am using a sdcard that I have been using since version 3.2 (class 10, 64gb).  In the deepest tests yesterday, I realized that things get worse with the option of active network play, it gets stuck when I enter all games.  Nintendo DS installation is not automatic.  the game gets stuck when entering the retroarch menu.  viewing the bios is taking too long to open.  when entering and exiting games, the menu quickly appears and sometimes crashes!  I did the tests in version 3.7ng, 3.8ng and 3.8arm.  in all of them the errors are the same.  the data is minimized when I leave the netplay option disabled (which for me is terrible) and I leave the collections disabled.  remembering that no game was placed other than those already in the image. @dm2912 @AlienatedDiego @shantigilbert

rochajunior, 2020-10-03:
I made some videos I will try to send you catches of errors, maybe I can help with something.

rochajunior, 2020-10-03:
https://streamable.com/fonof5
https://streamable.com/11n3mb
https://streamable.com/yl5607
https://streamable.com/dpqh3g
https://streamable.com/wqurg4

closed 2020-10-03 by shantigilbert