EmuELEC/EmuELEC #313: Commander Genius issues

Eyeball38, 2020-10-03:
# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**

1. Upon opening the port and reaching the launcher, the analog stick does not control the cursor making it very difficult to select a game.

2. Upon force closing the port a blank error log is displayed, this is also now happening when closing the port after returning to launcher.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to ports
2. Click on  commander genius 
3. Scroll down to game and launch
4. Force close, see error

**Expected behavior**

A functioning cursor to be able to easily select games and no error screen when closing the port

**EmuELEC version:**
 
- Version 3.8

**Device (please complete the following information):**
 
OGA black

**Additional context**

None to add

shantigilbert, 2020-10-03:
Hmm I think the issue here is that if I remember correctly, the cursor is controlled on the right analog stick, I will see if there is a way to change that for the OGA.

Eyeball38, 2020-10-03:
Thankyou

Cee123, 2020-10-05:
Interesting, I was thinking it was needing a mouse to be plugged into the USB port to be able to select the game.

Eyeball38, 2020-10-05:
If I very gently nudge the analog stick I am able to select different sections of the launcher, although it’s very difficult to see which part is highlighted. Once your on the game area, the same extremely gentle nudges allow to select the game and launch it. Does the connected mouse work?

Cee123, 2020-10-14:
I'm don't actually know. I could be wrong. But since I couldnt get the controls working for this, I assumed it was made for PC so probably needed a mouse or keyboard. But what Shanti said makes sense if it's the right analog stick that's why it's not working on OGA since it only has the left analog.

Eyeball38, 2020-10-14:
Im confident Shanti can find a way to re code the right to the left analog

closed 2026-02-01 by Langerz82