EmuELEC/EmuELEC #317: Weird controller issue - certain buttons/sticks not mapping and not recognized

Cee123, 2020-10-05:
**Describe the bug**
I'm having this weird issue where, depending on the controller, certain buttons become unrecognized and unmappable.

**To Reproduce**
So I have a few of these controllers I've purchased on eBay, I'm not really sure what brand they are and I know they're not great (I do have better controllers) but these are cheap and seem to work fine on all my other retro gaming setups.
[image]

For example on RetroArena, these work perfect, using the Xbox drivers (userspace drivers). Just wondering if there is any kind of an equivalent to this on EmuElec? If not, then it's no problem or anything and I'll stick with my Logitech controllers. Just thought I'd ask about it here.

**Expected behavior**
Seems to work on other retro setups.

**Screenshots**

**EmuELEC version:**
 All versions I've tried

**Device (please complete the following information):**
Happens on both Odroid N2 and my Amlogic Android Q Box 

**Additional context**
 Some of these controllers you have to hold down the home button first for about 10 seconds to switch the mode and then the light changes, and it recognizes it as an Xbox Controller (even though it isn't). Then it should work properly.

Cee123, 2020-10-16:
This is the specific driver (userspace driver) that fixes issues with xbox / xbox360 controllers on RetroArena that somehow also fixes problems with these controllers.
https://github.com/zerojay/xboxdrv.git

This script in RetroArena is the installer for it.
[xboxdrv.zip](https://github.com/EmuELEC/EmuELEC/files/5388633/xboxdrv.zip)

I might try and add this script to EmuElec and see if I can install the custom driver from there.

Cee123, 2020-10-16:
Nope, doesn't look like I have any way of installing this driver on EmuElec. Due to it being based on CoreElec, not even the command "apt-get" works in the terminal. Damn.

shantigilbert, 2020-10-16:
> Maybe you guys could add an optional script to install these userspace drivers for xbox/360 controllers.

why ? xbox controllers work fine, driver is baked in the kernel, why would we need a userspace one?  does that driver fixes the issue you are having on retroarena, have you tried it without it?

Cee123, 2020-10-16:
> > Maybe you guys could add an optional script to install these userspace drivers for xbox/360 controllers.
> 
> why ? xbox controllers work fine, driver is baked in the kernel, why would we need a userspace one? does that driver fixes the issue you are having on retroarena, have you tried it without it?

Well i tried again on retroarena with the userspace drivers. But it does exactly the same thing. The issue im now understanding is with the actual controllers themselves that i purchased on ebay not with the drivers. Sorry about that.

closed 2020-10-16 by Cee123