EmuELEC/EmuELEC #318: Function keys stopped working in Emulationstation EmuElec 3.8 OdroidGoAdvance

cederom, 2020-10-07:
**Describe the bug**
With version 3.8 function keys (start, stop, exit emulator, volume up/down, show keyboard, etc) stopped working in Emulationstation. They do still work in Retroarch.

**To Reproduce**
1. Put 3.8 image on a sd card and see that you cannot exit a game in Emulationstation.

**Expected behavior**
Function keys work as expected - I can exit a game to a menu, call keyboard, start, select, volume up/down.

**EmuELEC version:**
 - Version 3.8

**Device (please complete the following information):**
 - Odroid Go Advance Black (https://www.hardkernel.com/shop/odroid-go-advance-black-edition/)

shantigilbert, 2020-10-07:
Was this a clean install or upgrade? 

I have an OGA BE in my hand as I type this with a fresh install of 3.8 and hotkeys are working fine.

cederom, 2020-10-08:
Yes this is a clean install (dd image to sd first time setup partition resize etc). This happens on two different devices and two different cards.

cederom, 2020-10-08:
Hmm. I have reflashed the sd card image and on the raw image hotkeys seems to work fine.

I will try to trace the problem. Thank you for fast response :-)

cederom, 2020-10-08:
Okay, the cause of the problem seem to be: RetroArch / Online Updater / Update Controller Profiles.

Can someone please confirm? How can we update this update so it does not break things and also use it to fix up broken things already? :-)

TheLastProject, 2020-10-08:
I have heard Batocera disables this updating because it can break things. Maybe we should start disabling it too. What do you think, @shantigilbert?

cederom, 2020-10-08:
Thanks @TheLastProject :-) Maybe we can provide a valid Keymap configuration at RetroArch project for ODA - this way we could fix existing problem and extend RA supported platform list? I am not familiar with that project if its the right way and easy/possible to implement? :-)

shantigilbert, 2020-10-08:
Maybe you should have said that from the beginning of your post, you made it sound like you never changed anything...

Anyways, this is not going to get fixed, a lot of work has been done so that ES is the front-end, like I said on the other issue you opened, Lakka might be better suited for you. EmuELEC is based on Emulationstation.

cederom, 2020-10-09:
Hey @shantigilbert at first I thought that was default after updating system to 3.8.. sorry.. after careful examination it turned out to be RetroArch Controller Maps update.. I did not have this knowledge at first and none of you neither.. I am happy that we did that discovery and this may lead to a solution :-)

Thanks for the feedback on plans to abandon RetroArch.

shantigilbert, 2020-10-09:
Retroarch is not getting abandoned, it is used to run many emulators.

closed 2020-10-09 by shantigilbert

emuelecnoob, 2021-03-20:
Hi
I have problem with the retroarch hot key on Emuelec 4.0 as I set the hot key menu to hold start for 2 second on the retroarch but it doesn’t save the setting as i play some games and hot key menu not working. I then tried after update from retroarch and all buttons was inactive for all emulators.. apart from the emustation. I have to hold on power button to turn off on my Odroid Go Super. I don’t know what I have done wrong. Please advice

sky23h, 2021-10-01:
For any others that run into this, I figured it out without requiring a reformat.

I foolishly updated "Controller Profiles" too on my Odroid Go Advance clone and this breaks the controls, including the hotkeys to get into RetroArch and such.  I had an EmuElec backup that I restored, but it did not fix the controls. I was furious screaming my guts out.

What you have to do is from the EmuElec main menu, go to quick access menu, start retroarch, go to settings, input.  I had to re-map start, select, L3, & R3. Then from the input menu, go to Hotkeys, change the Menu Toggle controller combo to your choice, Check Hotkey enable is the button of your choice, and then set any other appropriate hotkey functions like load, save, restart, reset, etc. Be sure to save these configurations by going to RetroArch main menu, configuration file, save current configuration. My device is set to AutoSave config on exit but I still had to manually save for this to stick.