EmuELEC/EmuELEC #319: Wake up on Bluetooth controller connect

chucksmith, 2020-10-08:
**Is your feature request related to a problem? Please describe.**
Every TV box I know wakes up when its remote control is activated.

**Describe the solution you'd like**
When I turn on my 8BitDo SN30 Pro+ controller and it connects, I'd like my EmuELEC box to wake up immediately before I push another button on the gamepad for an ideal experience.

**Additional context**
I've had this issue on 3.7 as well as 3.8-TEST-09112020.

shantigilbert, 2020-10-08:
I have no idea what you are talking about... AFAIK, EmuELEC does not have a "sleep" function on SBC/TV Boxes. so what do you mean by "wake up"? 

Also TV Remotes are not Bluetooth.

chucksmith, 2020-10-10:
When I leave my EmuELEC TV box for a long time, I believe at least an hour, I return to a completely black screen. Then I connect my controller and have to push a button before it "wakes up". I'd like it to wake up immediately when I connect a controller in that state.

shantigilbert, 2020-10-10:
none of my boxes do that, if that is really the case then it has to be something box dependent maybe? Also I am sure the BT chip will turn off hence why it won't work.

chucksmith, 2020-10-12:
Oh wow, my BT controller actually connects and the Player 1 light lights up, but the screen doesn't show up until I press a button after it connects. Very weird! This is on the HK1 Box (S905x3).

closed 2020-10-16 by chucksmith