EmuELEC/EmuELEC #320: Selecting RetroArch or Emulationstation always ends up in Emulationstation ODA.

cederom, 2020-10-08:
**Describe the bug**
We have constant switches back from RetroArch to Emulationstation after exiting games in RetroArch. User selects either Emulationstation or RetroArch because they want to stick to that one solution not randomly jump from one to another. RetroArch seems more versatile and user friendly for kids. When selecting Boot the RetroArch we do not want to see the Emulationstation at all. With current jumping from one GUI to another users get confused and quickly abandon this solution. Maybe selecting the RetroArch as the default would be more feasible for newcomers?

EmuELEC 3.8 on ODROID Go Advance.

TheLastProject, 2020-10-08:
How exactly did you switch it? Did you go to settings in EmulationStation, then to EmuELEC settings and set "Start At Boot" to "RetroArch"?

I guess it could be considered a bug in both cases, but if it also happens with Start At Boot set to RetroArch I would consider it more severe.

cederom, 2020-10-08:
Yes exactly. As Eulationstation is the default GUI, I am setting the RetroArch in EmulationStation / EmuELEC settings so the GUI on boot is RetroArch.

Then randomly I am dropped back to Emulationstation even though the default is set to RetroArch and I am launching games from RetroArch..

TheLastProject, 2020-10-08:
Yup, can confirm the issue.

TheLastProject, 2020-10-08:
Okay the whole issue boils down to the fact that the "close game" hotkey actually is the "Quit RetroArch" hotkey when in libretro. Which means that it closes RetroArch completely even when playing a game.

And https://github.com/EmuELEC/EmuELEC/blob/e22939e635cfa789977b38449df548f110a91b89/packages/sx05re/libretro/retroarch/patches/retroarch-10-Exit_menu.patch#L29-L33 makes RetroArch start EmuELEC on closing RetroArch.

This behaviour seems fine if you switched to Emulationstation, but it definitely needs to change. The only thing I can imagine is somehow making the hotkey work differently when starting a game from RetroArch but not sure how...

shantigilbert, 2020-10-08:
If Retroarch is your main front-end, I recommend you use Lakka. EmuELEC will break if you use Retroarch like that, since a LOT of the functionality is removed from Retroarch to accommodate it for Emulationstation. Those options (Start Retroarch, boot to Retroarch) are there mostly for debug purposes, and probably should have been removed by now.

shantigilbert, 2021-01-19:
This is getting a second look, and I will probably come up with a fix in the coming versions to allow Retroarch to work standalone. 

but a huge issue is preventing this from happening right now as you can see here: https://github.com/EmuELEC/EmuELEC/issues/76

shantigilbert, 2022-02-16:
A fix has been made for #76 so I am looking into implementing this correctly.

shantigilbert, 2024-01-19:
I tested this, and I am not sure how this could change. The only solution I can think of is if Retroarch launches on boot, there should be no way to exit. Right now its made to exit back to Emulationstation because otherwise if you quit Retroarch and ES does not start it will just go to a black screen.

Langerz82, 2024-10-12:
@shantigilbert maybe have retroarch call a function and if retroarch is set to boot then just restart the retroarch service?