EmuELEC/EmuELEC #322: emuelec 3.8 n64 controls not working

lhschwelm, 2020-10-09:
Hello, I'm new here, I would like to inform you that the analog or directional controls are not working in most nintendo 64 games on emuelec 3.8 (the controls do not work on the mupen64plus_next or mupen64plus nor parallel_n64 emulators), could someone tell me if it is a known bug and how to solve?

Tks

NewAgeRipper, 2020-10-10:
Same here. Same issues with every controller I've tried. Most other systems work fine for me but N64 is a no go for me as well.

shantigilbert, 2020-10-10:
Disable digital to analog in Retroarch

Cee123, 2020-10-10:
Yeah same here, I'm having a problem with the analog stick not working properly on EmuElec.

NewAgeRipper, 2020-10-10:
Ah ok. Thanks. Will try that.

NewAgeRipper, 2020-10-10:
> Disable digital to analog in Retroarch

That did the trick after restoring scripts and what not in Danger zone. Now it's not wanting to download bezels even though the list of them came up.

TheLastProject, 2020-10-11:
Fixed by default in 5d956ddf68da049d904a2548d51a9e4685b972e7 by the way, so it should work out of the box the next release. If you need more help with the bezels still, please make a new issue. It's much easier to keep track of what is fixed and what isn't when we stick to one issue per subject.

closed 2020-10-11 by TheLastProject