EmuELEC/EmuELEC #323: Cdtech 4761743 WiFi support

yaway, 2020-10-10:
I'v got a box with Cdtech 4761743 chip (WiFi works on some other customized android OS). It's wil be cool if support for the WiFi and Bluetooth to be added!

[image]

shantigilbert, 2020-10-11:
Unless you provide the driver that is confirmed working, I have no device (nor have I ever even heard of this brand) that uses this chip

shantigilbert, 2021-01-19:
Closing this, if you get more information on the driver, please feel free to reopen

closed 2021-01-19 by shantigilbert

eric4gith, 2021-03-05:
I opened the case of my X96 Max Plus and found the same chip Cdtech 4761743. It's interesting.
According to some info I found on pda.ru (transalted to English from Russian, cited below):
https://jbxpgt7fabdpg2ikfwmq5nxn7i--4pda-ru.translate.goog/forum/index.php?showtopic=1013103&st=140#entry103421297
-----------------------------------------------------------------------------------------------------
I'm just going to buy this console while I'm studying the topic. I downloaded the firmware just in case.
But I searched a little on the Internet about cdtech 4761743. On the site
http://www.cdtech.cn
there is information only on cdtech 4761741 on Chipsete ： QCA6174A-1.
Judging by the name of cdtech and QCA6174A-1, I can assume that cdtech 4761743 may have Chipsete ： QCA6174A-3.
on the website
www.qualcomm.com
about this Chipsete they write what it is: High-performance dual-band Wi-Fi 802.11ac 2x2 with MU-MIMO and Bluetooth 5.0 radio modules in a single-chip solution and it has
three versions:
QCA6174A-1: Supports PCIe 2.1 Low Energy (L1 Subsystem) for WLAN and UART / PCM for Bluetooth
QCA6174A-3: Supports SDIO 3.0 Low Energy for WLAN and UART / PCM for Bluetooth
QCA6174A-5: Supports PCIe 2.1 Low Energy (with L1 subsystem) for WLAN and USB 1.1 for Bluetooth.
We already have a description of the characteristics of the X96 Max Plus on one site: 
-------------------------------------------------------------------------------------------------------------

FYI.

kakaxp, 2021-05-20:
> Closing this, if you get more information on the driver, please feel free to reopen

I have the same problem...

Searching on google I saw that the niglthy version of coreelec already has the drivers for wifi and bt of this box.

Would it be possible to get the drivers inside the corelec?

Thanks

Link: https://relkai.coreelec.org/CoreELEC-Amlogic-ng.arm-19.2-Matrix_nightly_20210520-Generic.img.gz

shantigilbert, 2021-05-20:
429c8bb6bfa65565488682de5b666bae5223b4f6