EmuELEC/EmuELEC #324: Emuelec 3.8 - Reicast OLD not working/Working Slow on S905x

mymo1982, 2020-10-10:
As per subject, some Dreamcast games (i.e. Crazy Taxi 1 and 2) won't start when selected Reicast Old emultator, others (i.e. Jet Set Radio) have a very poor framerate. They do work with Flycast but performaces are not on par with Reicast by far, both in terms of framerate and resolution. 
All the games I mentioned worked like a charm on version 3.6. 
Also, I regret to say that menu navigation has become stuttery on ver. 3.8.
My system is a MXQ mounting an AML S905x chipset and 1 GB of Ram.

shantigilbert, 2020-10-10:
First, please don't mix issues, if you have issues affecting different stuff open one per issue.

Second, is this a fresh install? or an upgrade? did you do the proper upgrade method explained in the Wiki? 
Same roms? what framerate were you getting on 3.6 for the games you mention?

Reicast old has not been updated (hence why is called old) in any shape since it was introduced.

mymo1982, 2020-10-10:
I'm sorry that I mixed the issues, the menu navigation problem is a minor one, I wouldn't have opened a topic only for that. In any case, let's not consider that now.

I had upgraded from 3.6 to 3.8. I followed this guide 

https://github.com/EmuELEC/EmuELEC/wiki/How-to-update

Before updating I had also restored the system to default settings.

I am not sure how to get a framerate counter on Reicast old since it's not a libreto core, but at a glance I am sure it's way more than 30 fps at fhd on 3.6. On 3.8 I'd say it's 18/20 fps and choppy sound, and many games won't even start.

I know Reicast old has lots of issues since it's not updated, but it's the only one that gives a decent performance at high resolution on S905x. Recast new just doesn't work very well, and Flycast is ok but Retroarch overhead is heavy and affects its final output.

shantigilbert, 2020-10-11:
Can you do me a favor and test the aarch64 test version? (from here https://github.com/EmuELEC/EmuELEC-tests/releases)I am using s905x with Aarch64 right now, and I can play crazy taxi 1 very well (and many other DC abd atomiswave games), no slowdowns at all. Wonder if the architecture has anything to do with it?

mymo1982, 2020-10-11:
So I tried the aarch64 version. Flycast runs at 30FPS at 800x600 and at 15/18FPS at 1600x1200 (which I guess is the standard resolution in ReicastOld).

shantigilbert, 2020-10-11:
Hmm I think you are asking way too much, Reicast did not have 1600x1200 internal resolution, that is not doable on a S905x, you need to run it at NATIVE resolution (640x480) and it most definitely does not run at 30 fps on my test if you use native resolution.

mymo1982, 2020-10-11:
You are right, I am asking too much because native resolution is much lower, but Reicast can handle higher resolutions. Below you can find attached some pictures I took with both Flycast and Reicast OLD. I had to take it with my phone because if I took a screenshot it would not include the framerate count. In any case:

Here it's Flycast on 3.8 aarch64, running at 1600x1200 with a 14.58 framerate
https://1drv.ms/u/s!AvTsunqJvbinh16TrwNj5SHN4f-f

Again, Flycast on 3.8 aarch64, running at 800x600 with a 31.84 framerate
https://1drv.ms/u/s!AvTsunqJvbinh19ePYdbRuLdncKs

As for Reicast, I don't know how to check resolution or framerate. Anyway from the pictures I attach below one can tell it is not native resolution, but it must be around 1600x1200 
https://1drv.ms/u/s!AvTsunqJvbinh2EFmVvClPs1bu_p
https://1drv.ms/u/s!AvTsunqJvbinh2CobiCf-zXQX1cI

As for the framerate in Reicast, here I have a short video where it shows it is not stuttery at all, even at a higher resolution
https://1drv.ms/v/s!AvTsunqJvbinh10_S3o5ScmOmCHC

Of course Reicast old is not perfect. Many glitches are present, many games are not compatible, and it is not supported anymore. But when it works there's no comparison, it is way better than Flycast.

shantigilbert, 2020-10-11:
Sorry but again no, that is just impossible for this SOC. But if you feel like that is what is it, then I would recommend you stay at 3.6, since Reicast is going to be removed on aarch64 from v4 (it does not compile and I do not feel like fixing it on my own) and no 32bit version will be available.

closed 2020-10-11 by shantigilbert

mymo1982, 2020-10-11:
Well, it is very rude to shut down a topic just out of spite when I spent 
time to take pictures and videos to prove that I wasn't a liar.
My point had nothing to do with Emuelec but only with cores on which you 
have no control on, so I wasn't talking bad of your product at all.

Hmm I think you are asking way too much, Reicast did not have 1600x1200 
internal resolution, that is not doable on a S905x, you need to run it at 
NATIVE resolution (640x480) and it most definitely does not run at 30 fps 
on my test if you use native resolution.

shantigilbert, 2020-10-11:
Why was it rude? I gave you the answer, why have it open just because you don't like the answer? photos don't change the reality.

mymo1982, 2020-10-11:
You weren't obliged to give me any answers, I just pointed out that an 
emulator is not compatible with your product: You could have  said it 
wasn't supported anymore and it would have ended there.
Photo (and the video) are clear proof that my statements had basis and I 
wasn't trying to fool you or waste your time. If you didn't believe it, you 
could've asked for further proof instead of acting out of impulse.

shantigilbert, 2020-10-11:
Dude, just stop, nobody said you were lying, I know the emulator is not compatible, I removed it my self. The emulator has not been updated in for ever, nothing changed in regards to versions, but if you think 3.6 runs better for your (very old) SOC then by all means use that! I am just stating facts that it is IMPOSSIBLE to run DC at those resolutions using this (again, very old) SOC with 1GB of ram none the less, just because you took 10 seconds to record photos and video does not make it any more or less true, you even said it yourself, you have no way of knowing what resolution or frame rate you get with Reicast, why assume its that? If you are getting salty because I am telling you this, then its on you, no need to try and act all mighty. No further proof is needed, as I know that even the N2+ struggles with DC on other resolutions than the native.

mymo1982, 2020-10-11:
Dear Shantigilbert,
In my everyday life I'm a lawyer, so you're pleased to address as "dude" a 
member of you family or an associate of yours, not me.
You didn't clearly state the Reicast old is not supported: it's actually 
still present in version 3.8, how can a user know it's not supposed to be used?
You also assume I am dying to use version 3.8: the truth is, as soon as I 
found out it didn't suit me, I went back to version 3.6, and I opened the 
topic on GitHub just to let you know that there was a bug. I tried the 
aarch64 version and reported on it just as per your suggestion.
It's true that I can't prove what resolution and framerate Reicast runs at, 
but the video is evidence that it does run better than Flycast, which is a 
widely known fact, as pointed out by many users in many forums.
Anyway, we both spent too long on a non existent issues, and as far as I'm 
concerned this ends here.