EmuELEC/EmuELEC #325: Is it possible to make it possible to turn down the volume of the preview video for games

Sidordroid, 2020-10-13:
Dear developers, thanks for the cool OS. But there are a few questions 1) Maybe I will repeat myself, but you can add Kodi, if it can't be added, then it didn't hurt to add another multimedia station. 2) Is it possible to make it possible to turn down the volume of the preview video for games - they are too loud and cannot be turned down video volume

shantigilbert, 2020-10-13:
No, this is not a full fledge os, it is meant ONLY for emulation, but a small media player is already present not for 4k videos or anything above 1080p.

shantigilbert, 2020-10-13:
Also please read the issue templates they are there for a reason

Sidordroid, 2020-10-14:
> No, this is not a full fledge os, it is meant ONLY for emulation, but a small media player is already present not for 4k videos or anything above 1080p.

And how to watch video files and listen to music on EmuElec. Please click on the link to the instruction. Thank you

Sidordroid, 2020-10-14:
> > No, this is not a full fledge os, it is meant ONLY for emulation, but a small media player is already present not for 4k videos or anything above 1080p.
> 
> And how to watch video files and listen to music on EmuElec. Please click on the link to the instruction. Thank you

closed 2020-10-14 by Sidordroid

Sidordroid, 2020-11-26:
I still don't know how to use the media player in Emuelec. Stick my nose in the instructions please

shantigilbert, 2020-11-26:
place your videos in /storage/roms/mplayer and it should appear in ES after a restart

Sidordroid, 2020-11-26:
Thank you. Can you listen to music tracks, for example mp3? Where to put them?

shantigilbert, 2020-11-26:
not while playing, only as background music for ES `/storage/roms/BGM`