EmuELEC/EmuELEC #326: Installing EmuELEC on a RK2020

fabulousrice, 2020-10-14:
I would like to install EmuELEC on a RK2020, but I am not sure which is the right ".img" file to flash (I use Balena Etcher for Mac). Please let me know which one is the correct link! I am also not much of a programmer. Thanks!

TheLastProject, 2020-10-14:
I'm sorry, but we don't support the RK2020 due to copyright infringement issues. See https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices.

closed 2020-10-14 by TheLastProject

AlienatedDiego, 2020-10-14:
Unfortunately EmuELEC does not offer support for this device, you are using an unofficial fork, you should contact the manufacturer to provide you with compatible software

fabulousrice, 2020-10-14:
> I'm sorry, but we don't support the RK2020 due to copyright infringement issues. See https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices.

Oh, because piracy is bad, ok, get it.

Cee123, 2020-11-09:
@fabulousrice EmuElec works on the RK2020, as I found from experience. The hardware is basically the same. I've tried it out on mine, and the you image you'd need is the one for the Odroid Go Advance. I can use the same SD for my Odroid Go Advance and my RK2020 and it works in both.

I had actually purchased one of these first because no matter what I did at the time, I just couldn't get my hands on an OGA. The website was constantly out of stock and frustratingly the two I purchased on eBay went missing in the mail. Fortunately I finally managed to obtain one, now I have one of each. They're pretty much the same. The RK2020 feels like a slightly nicer design, but I also like the extra customizable buttons on the OGA.