EmuELEC/EmuELEC #327: tv box tx3 mini 2/16g - emuelec 3.8

kaderr09, 2020-10-15:
hello
i have red all the issues but have not found this one.
i have Tx3 mini 2/16gb with s905w ,i burned the image for several emuelec versions till the last one 3.8
when i start the tv box The emuelec screen appears, creates partition, restarts then the image seems too big! after that there is a black screen for ever. i have a video that describes the situation.
youtube video :  https://www.youtube.com/watch?v=4AZJMexG49E&t=35s
any help would be appreciated 
thank you for your support

shantigilbert, 2020-10-15:
Go into android, set resolution to 1080p,revoot to emuelec

kaderr09, 2020-10-16:
> Go into android, set resolution to 1080p, reboot to emuelec

thank you to the bone it works
so sorry to ask another question about the same tv box and the same memory card, but with: Galisteo_Cobalto_Emuelec_32GB(v3.5).img ..... when i boot the tv box it freezes at the starting page of emuelec 
i have waited for about 10 mins but nothing happend

shantigilbert, 2020-10-16:
No pirated images supported

closed 2020-10-16 by shantigilbert