EmuELEC/EmuELEC #328: Video screen saver mute does not work

chucksmith, 2020-10-16:
**Describe the bug**
I'd like to use the video screensaver without sound, but even when I go into Screensaver Settings and turn on Mute Video Audio, the screensaver still plays videos with sound.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Screensaver Settings and choose Video.
2. Turn on Mute Video Audio.
3. Launch Screensaver.
4. Hear audio playing with random videos.

**Expected behavior**
I expect to see random game videos without sound.

**EmuELEC version:**
 - Version 3.8

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: S095x3
 - Model: HK1 Box

shantigilbert, 2020-10-22:
I just tested this and all of my videos play muted... on the video screen and on the video screen saver, can you post the format of one of your videos?

chucksmith, 2020-10-24:
This was an issue for me in 3.7 and 3.8, but appears to be fixed in 3.9, so I'm closing this ticket.

closed 2020-10-24 by chucksmith