EmuELEC/EmuELEC #329: Does not download kernels and not all kernels are available for download?

Sidordroid, 2020-10-16:
I am using EmuElec on an amlogic s905x3 tv box and I have identified some shortcomings in the work of Retroarch. Does not download kernels and not all kernels are available for download? When trying to forcibly copy my files from another Retroarch, it does not see them. Is it write-protected? Why is it cut? to put in EmuElec, copy your full-fledged and not trimmed Retroarch?

shantigilbert, 2020-10-16:
Not sure what you mean by Kernels, if you mean cores then, please do not download them it will break your system, the cores included are the ONLY ones supported.

closed 2020-10-16 by shantigilbert

Sidordroid, 2020-10-16:
In EmuElec in the Dreamcast emulator, the speed is good, but with fast loading and saving, it gives this thing. And the 3DO emulator in EmuElec lags a lot
[image]

Sidordroid, 2020-10-16:
> In EmuElec in the Dreamcast emulator, the speed is good, but with fast loading and saving, it gives this thing. And the 3DO emulator in EmuElec lags a lot
> [image]

Sidordroid, 2020-10-16:
In EmuElec on the Dreamcast emulator, the speed is good, but it gives this with fast saving and fast loading. And the Panasonic 3DO emulator lags a lot
[image]