EmuELEC/EmuELEC #342: psp roms do not work

kaderr09, 2020-10-29:
hello
i have the tx3 mini s905 with muelec 3.9 . everything work great but when i add psp roms and try to play them they crash i think it has relation with the bios
wich bios file fits emuelec if you have a solution
thank you for all what you do to improve the project

shantigilbert, 2020-10-29:
Really? Can you just follow the TEMPLATE that is not enough information!

closed 2020-10-29 by shantigilbert

kaderr09, 2020-10-30:
> Really? Can you just follow the TEMPLATE, that is not enough information!
> 
> PPSSPPP does not need any bios, so unless you give way more information than this, no help will be given, that is why the issue template exists.

https://www.youtube.com/watch?v=ang6-QG-jJQ
the video explains the issue