EmuELEC/EmuELEC #343: how to make psp roms work

kaderr09, 2020-10-30:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!
hello i have a tx3 mini tv box wit s905 ,everything is ok excepting psp roms 
thy do not work . and this is a link of a video witch describes the situation  
https://www.youtube.com/watch?v=ang6-QG-jJQ
 any suggestion please

shantigilbert, 2020-10-30:
You do not have a S905, that is a s912, PPSSPP stand alone does not work on S912, use the libretro one. and Please stop posting this!

closed 2020-10-30 by shantigilbert

kaderr09, 2020-10-30:
> You do not have a S905, that is a s912, PPSSPP stand alone does not work on S912, use the libretro one. and Please stop posting this!

sorry i have posted it again because the previous posts do not have enough information 
i thought tx3 mini is s905 from browsing it at the official site 
. CPU: Amlogic S905W Quad-Core 64bit Cortex-A53 1.5GHz
thank you for your patience and so sorry for re-posting again

shantigilbert, 2020-10-30:
They sell a gazillion different "TX3" devices, none of them have the same components, yours is a S912.