EmuELEC/EmuELEC #344: Youtube-dl downloading fails while building addon package

lzivadinovic, 2020-11-02:
**Describe the bug**
Cant build emuelec plugin for Amlogic-ng; Problem is youtube-dl DCMA take down.

**To Reproduce**
Steps to reproduce the behavior:
1. `./emuelec-addon.sh`  fails

```bash
Cannot get youtube-dl sources : https://github.com/ytdl-org/youtube-dl/releases/download/2020.06.16.1/youtube-dl
Try later!
```
(and a lot of 2020-11-02 13:45:10 ERROR 404: Not Found.)

**Expected behavior**
Build should succeed 

**EmuELEC version:**
 - Version latest master (a2a3429)

**Device**
Just building plugin for amlogic-ng platform (i've commented out amlogic only in build script). Im using ODROID N2

**Additional context**
I was able to "fix" build with following update, but i have not tested.

```bash
diff --git a/packages/sx05re/tools/video/youtube-dl/package.mk b/packages/sx05re/tools/video/youtube-dl/package.mk
index 06816c6dc0..749c819cd7 100644
--- a/packages/sx05re/tools/video/youtube-dl/package.mk
+++ b/packages/sx05re/tools/video/youtube-dl/package.mk
@@ -2,21 +2,23 @@
 # Copyright (C) 2020-present Shanti Gilbert (https://github.com/shantigilbert)
 
 PKG_NAME="youtube-dl"
-PKG_VERSION="2020.06.16.1"
-PKG_SHA256="3062fcc8e511cec6ebbf58c87894bb64273d4668cc584439e4c722b76ff0c9eb"
+PKG_VERSION="2020.11.01.1"
+PKG_SHA256="7a62187bc9f91af42eb5c9c95652e350170a72dbbdac8c7d71a6aeb4885a901a"
 PKG_LICENSE="The Unlicense"
-PKG_SITE="https://github.com/ytdl-org/youtube-dl"
-PKG_URL="${PKG_SITE}/releases/download/${PKG_VERSION}/youtube-dl"
+PKG_SITE="https://youtube-dl.org"
+PKG_URL="${PKG_SITE}/downloads/latest/${PKG_NAME}-${PKG_VERSION}.tar.gz"
 PKG_DEPENDS_TARGET="toolchain"
 PKG_LONGDESC="Command-line program to download videos from YouTube.com and other video sites"
 PKG_TOOLCHAIN="manual"
 
 unpack() {
 :
+
 }
 
 makeinstall_target() {
 mkdir -p ${INSTALL}/usr/bin
-cp -rf ${SOURCES}/${PKG_NAME}/${PKG_NAME}-${PKG_VERSION}.youtube-dl ${INSTALL}/usr/bin/youtube-dl
+tar -zxvf ${SOURCES}/${PKG_NAME}/${PKG_NAME}-${PKG_VERSION}.tar.gz youtube-dl/youtube-dl --strip-components=1
+cp -rf ${SOURCES}/${PKG_NAME}/youtube-dl ${INSTALL}/usr/bin/youtube-dl
 chmod +x ${INSTALL}/usr/bin/youtube-dl
 }
```

shantigilbert, 2020-11-02:
You can fix it with this: https://github.com/EmuELEC/EmuELEC/pull/338

closed 2020-11-02 by shantigilbert

lzivadinovic, 2020-11-02:
Ive missed this. Thanks for pointing out this PR.