EmuELEC/EmuELEC #346: I HAVE AN ANDROID BOX AND I CANT FIND WHAT DEVICE TREE I USE PLEASE HELP ME

sheza0890, 2020-11-07:
PLEASE SOMEBODY HELP ME, I HAVE BEEN TRYING FOR SO LONG BUT I AM NEW TO THIS. I WANT TO BE ABLE TO PLAY ALL THE OLDER GAMES BUT I JUST DONT KNOW WHAT DEVICE TREE I USE.
[image]
[image]

shantigilbert, 2020-11-07:
Please read the wiki https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install

closed 2020-11-07 by shantigilbert

sheza0890, 2020-11-07:
I did read it. I don’t really understand it all

TheLastProject, 2020-11-07:
The place where you bought it will probably tell you what processor is in it, assuming it's Amlogic. I did a quick Google for T95 and it said "Processor: S905 quad-core cortex-A53", so that's S905. So according to https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install#how-to-install that means you need `EmuELEC-Amlogic.arm-x.x.x-generic.img.gz`

dm2912, 2020-11-08:
T95 with 4tgb ram, I think its Allwinner H616

this one? 

https://www.amazon.co.uk/Android-Allwinner-Quad-Core-Cortex-A53-Ethernet/dp/B08DRZM374/ref=sr_1_5?dchild=1&keywords=t95&qid=1604796970&sr=8-5

In which case its not AmLogic

programmo, 2021-05-05:
> Please read the wiki https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install

Hi me too same problem... 
Can you tell me or others how to pass the problem ?
Thank you

dm2912, 2021-05-05:
> > Please read the wiki https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install
> 
> Hi me too same problem...
> Can you tell me or others how to pass the problem ?
> Thank you

you have the same problem as in you have that box with an allwinner cpu? (In which case its not compatible with emuelec)

Or you have a totally different box and we are supposed to just guess what device you have? :D

programmo, 2021-05-06:
> you have the same problem as in you have that box with an allwinner cpu? (In which case its not compatible with emuelec)
>
> Or you have a totally different box and we are supposed to just guess what [...]

Hi thank for reply.
It is the same device with allwinner CPU so what I do with this device ?
It is planned to do a version for this CPU ?
So what to do with this device  ?
To trash ?
Thank 
Have a nice day.

dm2912, 2021-05-06:
> Hi thank for reply.
> It is the same device with allwinner CPU so what I do with this device ?
> It is planned to do a version for this CPU ?
> So what to do with this device ?
> To trash ?
> Thank
> Have a [...]

emuelec is specifically for amlogic devices.

RaidenZala, 2024-01-09:
Hello. I'm trying to install emuelec but cannot find the right image. My device is Movie Cube F400 by EMTEC. Please help.

SoC: AMLogic Meson6 AML8726-MX
Core Architecture: 2x ARM Cortex-A9 @ 1200 MHz
Instruction Set: 32-bit ARMv7
CPU Revision: r3p0
Board: f400
Device: f400
Platform: meson6
Product: f400
Installed RAM: 1GB

Thanks in advance