EmuELEC/EmuELEC #349: Screen position. Emuelec 3.9. X96 mini 2/16 via RCA.

bennington111, 2020-11-13:
Hi. I use Emuelec 3.9 on x96 mini via RCA using this guide: https://github.com/EmuELEC/EmuELEC/wiki/Salida-RCA. 
Screen shifted left. 
How to change screen position? 
Which file contains the screen position settings?
[image]
[image]

shantigilbert, 2020-11-16:
not easy to do, but follow this issue. 

https://github.com/EmuELEC/EmuELEC/issues/61

shantigilbert, 2020-11-25:
No reply for 10 days, will assume fixed, otherwise please feel free to reopen this issue.

closed 2020-11-25 by shantigilbert