EmuELEC/EmuELEC #353: No sound in game emuelec 4.0-TEST-11162020 on x96 mini 2/16 (s905w).

bennington111, 2020-11-17:
No sound in game emuelec 4.0-TEST-11162020 on x96 mini 2/16 (s905w) hdmi .

AlienatedDiego, 2020-11-17:
Thanks, we suspect some problems with test 4.0 test for s905 family
Please check if any of the unconfirmed issues are reproduced:

Crashes in E.S when navigating between systems

sound problems in some cores like mame, flycast genesis + or snes9x; in others like pcsx_rearmed or ppsspp it seems to work

sound problems in E.S (videos)

unstable or running cores too fast (Mupen 64, flycast)

Remember that it is only a first test to identify problems.

shantigilbert, 2020-11-18:
Thanks I can confirm it does not work in Retroarch, I will investigate.

shantigilbert, 2020-11-19:
Uploaded a fixed image: https://github.com/EmuELEC/EmuELEC-tests/releases/tag/v4.0-test

closed 2020-11-19 by shantigilbert