EmuELEC/EmuELEC #354: [QUESTION] What script is re-recreating the directories in /storage/roms on every boot?

vanfanel, 2020-11-22:
Hi!

I am trying to get EmuELEC as I want it: no EmulationStation, just RetroArch, with my own directories for ROMs in /storage/roms/.

In order to get there, I have already removed the emuelec-autostart.service, that in turn runs `/storage/.config/autostart.sh`.
SO autostart.sh should NOT be running on my custom system.

However, the directories inside /storage/roms are still being recreated in every boot and I can't determine WHAT is recreating them.
I have also disabled the emustation service, just in case emustation would be creating them when running emulationstation somehow.

Any help is welcome!

shantigilbert, 2020-11-22:
https://github.com/EmuELEC/EmuELEC/blob/master/packages/sx05re/emuelec/tmpfiles.d/emuelec-dirs.conf

You could recompile the distro without that file or my suggestion would be to just create a symlink to which ever folder you want your roms in. 

Also You should try Lakka, it is exactly as you want.

vanfanel, 2020-11-22:
> https://github.com/EmuELEC/EmuELEC/blob/master/packages/sx05re/emuelec/tmpfiles.d/emuelec-dirs.conf
> 
> You could recompile the distro without that file or my suggestion would be to just create a symlink to which ever folder you want your roms in.
> 
> Also You should try Lakka, it is exactly as you want.

Oh, thanks a lot! What program or script is using this emuelec-dirs.conf file?
No way to override it's contents without rebuilding the system?

Also, yes, Lakka is what I want, but has some problems in RG351P, whene I am using Emuelec-derivative 351elec.

shantigilbert, 2020-11-22:
> No way to override it's contents without rebuilding the system?

Not that I know of, sorry

vanfanel, 2020-11-22:
@shantigilbert Do you know what program is using this .conf? Out of curiosity, really.

shantigilbert, 2020-11-23:
systemd

vanfanel, 2020-11-23:
> systemd

Oh, ok. Thanks!

closed 2020-11-23 by vanfanel

shantigilbert, 2020-11-23:
https://github.com/EmuELEC/EmuELEC/blob/master/packages/sysutils/systemd/config/tmpfiles.d/README