EmuELEC/EmuELEC #355: Resolution bug on minix neo u9 S912

greekuser, 2020-11-24:
My tv is a Samsung 4k. if i choose from settings 3840x2160p, after the reboot i see the emuelec start logo and then a green screen! I have to remove power OR i can shutdown the system from the menu by doing blind navigation (so the system is not frozen). Then the only way to change the resolution to 1080p is by changing the "emuelec.conf" file or by blind navigation again. my problem is not that, i can live with 1080p. 
The real bug is a mix of green screen, a totally overscanned menu and menu overlaying tha same menu. if i am lucky i can navigate to restart emulestation where the resolution fixes. Switching to retroarch and back doent have any problems. Getting in and out of games is ok too. I have tested and it is the same on both emuelec 3.8 and 3.9!
I am using the minix neo u9 dtb. If i use other s912 dtb i get only a black screen, doesnt boot at all. Also tested on a native 1080p tv.

Steps to reproduce the behavior:
1. 'start any game that needs bios which i dont have. i get an error screen showing what is missing'
2. 'click cancel to exit back to emulestation'
3. Voila...i cant see a clean screen when going from the error screen back to main menu. The resolution or something else gets confused. I dont even think if such thing is logged. Please tell me what log, anything do you want upload and where to find it.

**EmuELEC version:**
 - Version  tested on 3.8, 3.9

**Device
 - SOC: s912h
 - Brand: minix
 - Model  neo u9

shantigilbert, 2020-11-24:
Problem is, EmuELEC is not really meant to be used at 4k, and the issue you mention might come from the fact that Android sets 4k on the resolution, so when you reboot the device with EmuELEC even tho emuelec is st to 1080p the Android setting is being used. 
For testing, you can set the resolution to 1080 in Android, the reboot to EmuELEC, if the problem is not there, then it means that this is the issue and unfortunately I have no real idea how to fix that.

greekuser, 2020-11-24:
Thanks for your reply. I have never thought that android settings would matter because android isnt even started. Thats the point of booting different software and not using kodi addon! So i changed the resolution inside android to 1080p and unfortunately the bug still occurs in EmuELEC.
I know that it is a free product and by no means i ask/demand a solution. I also read that s912 isn't the developer's main box with poor driver support. I am just saying my bug and i am perfectly happy to help you investigate more and fix it. I am guessing it must be an easy fix??

ibraheeemz, 2020-11-24:
> Thanks for your reply. I have never thought that android settings would matter because android isnt even started. Thats the point of booting different software and not using kodi addon! So i changed [...]

I also have the same setup, android box with s912 and a 4K TV, and after initial boot and setting the controllers, once I started a game, I see the green flickering screen as well. I have another TV with 1080p display only, I will check on that also and let you know if it worked without problems.

greekuser, 2020-11-24:
> I also have the same setup, android box with s912 and a 4K TV, and after initial boot and setting the controllers, once I started a game, I see the green flickering screen as well. I have another TV with 1080p display only, I will check on that also and let you know if it worked without problems.
> 
> Update: I am still seeing the green screen even on the other TV, and scattered graphics after next reboot. resolution seems to be changed and all the options are too big now...

That's too bad that you cant even play. At least on the Minix neo u9 i can play normally.

shantigilbert, 2020-11-24:
Unfortunately it does not work as you think it does (and not all TV Boxes are made the same)  this is my understanding:

1.- Tv box loads the internal boot settings (manufeacturer settings, and where most Android settings are stored)
2.- It will load any uboot either on the SD, USB or internal (external OS)
3.- boots whatever OS  

The 1st part can't be changed AFAIK. I could be very wrong about this, and feel free to correct me if someone does know for sure. 

Anyways, unfortunately it is not an easy fix, I have no such device to even play with it, and even if I did, I do not have a 4k TV only a 720p and a 1080p TV.

you can try this information from the WiKi and see if that helps: https://github.com/EmuELEC/EmuELEC/wiki/Common-Questions#how-to-change-resolution-i-get-black-screen-or-my-tv-is-not-working-with-your-image

But honestly one of the reasons s912 is not going to be supported anymore is because of how hacky the GPU drivers are which cause so many weird issues on different boxes (my only s912 box works perfectly fine on my TVs for example) and my knowledge about dealing with the s912 issues related to GPU  is close to zero.

greekuser, 2020-11-24:
I know about s912 issues, i didnt buy it for EmuELEC, i just had it(it is a perfect 4k hdr streamer with kodi + hd audio passthrough)! 
About the wiki, it says only how to change resolution if i see a black screen. So in my STORAGE partition i can edit "emuelec.conf" , but in my EMUELEC partition(3.8 and 3.9) there is no "config.ini", not hidden, nothing. Anyway i created a new file just in case and wrote inside what i saw  at wiki. I tried every single resolution from 640x480p up to 1080p . Lower sd resolutions didnt work at all from the beginning. For example it shows only the 640x480 part of the full hd image. And of course the same thing happened when exiting the EmuELEC error log. Green flickering. I concluded at 1080p which is the best with that bug.
I assume when you start the game ,  retroarch first loads, then core(emulator) loads, then game loads?!? Then if the game is not ok, EmuELEC error log shows up. For a second, just before error log shows up, the green, flickered, messed up screen appears, so the bug has already happened before loading or while loading the error log or when a bad game is run. I would look at that part of code if i knew how. But my programming knowledge is limited to C only and basic stuff with avr/pic. I know nothing about ARM, hdmi, kernels and  all that cool stuff :/

ibraheeemz, 2020-11-25:
> I know about s912 issues, i didnt buy it for EmuELEC, i just had it(it is a perfect 4k hdr streamer with kodi + hd audio passthrough)!
> About the wiki, it says only how to change resolution if i see [...]

exactly the same scenario and observations for me too. didn't find the conf file in EMUELEC partition and also in the conf file in storage partition, the setting "hdmimode" is not readily available. I was thinking to use AV in/out on the box and then on the EMUELEC and maybe it will have better performance or display. Actually I am doing all this for my kid. The box performance itself for the TV part is great. Plays 4K videos like a charm. So the drivers and box capability is not an issue. Having some experience with bash scripts, I could try to play around with different config files but I haven't played with drivers/kernels ever. I understand that it would be too much to ask from the dev to debug any such case but would be happy to do so myself if I could.

shantigilbert, 2020-11-25:
> * 'start any game that needs bios which i dont have. i get an error screen showing what is missing'
> 
> * 'click cancel to exit back to emulestation'
> 
> * Voila...i cant see a clean screen when going from the error screen back to main menu. The resolution or something else gets confused. I dont even think if such thing is logged. Please tell me what log, anything do you want upload and where to find it.

The only way I could think of is by disabling fbterm completely. 

Open the file `/emuelec/scripts/fbterm.sh` either by SSH or by Samba share, then add this `exit 0` on line 5
If that doesn't work you can also add `exit 0` on line 11 in `/emuelec/scripts/emuelec-utils`

keep in mind bios errors, and other features that need fbterm will no longer work.

ibraheeemz, 2020-11-25:
So I wanted to know, if I were to purchase any android box to only install EMUELEC specifically, which one is the best supported box and with what specifications? And is EMUELEC installed/supported on other platforms as well other than android boxes? Like Raspberry pi, laptops etc... ?

shantigilbert, 2020-11-25:
Please read the wiki, it explains this.

greekuser, 2020-11-25:
> exactly the same scenario and observations for me too. didn't find the conf file in EMUELEC partition and also in the conf file in storage partition, the setting "hdmimode" is not readily available. I was thinking to use AV in/out on the box and then on the EMUELEC and maybe it will have better performance or display. Actually I am doing all this for my kid. The box performance itself for the TV part is great. Plays 4K videos like a charm. So the drivers and box capability is not an issue. Having some experience with bash scripts, I could try to play around with different config files but I haven't played with drivers/kernels ever. I understand that it would be too much to ask from the dev to debug any such case but would be happy to do so myself if I could.
> 
> Update:
> I tested using AV cable. My Android box works fine while EMUELEC doesn't show any display. Switching to hdmi, without even rebooting gives the display.

greekuser, 2020-11-25:
> The only way I could think of is by disabling fbterm completely.
>
> Open the file `/emuelec/scripts/fbterm.sh` either by SSH or by Samba share, then add this `exit 0` on line 5
> If that doesn't work [...]

I really thank you for your time to even think of a solution. So i SSHed and done your first proposal. It worked like a charm. After the EmuELEC game loading screen, it goes back to game list, no green flickering, no mixed up resolution. No error log of course, but we can live without that. I am not even gonna try the second proposal. 
But i have to ask you one last thing, If you have time to think. if we need the error log, what to i need to edit, for when i press cancel(in the error log) the emulstation(not system) restarts automatically and not just exit to gamelist.
Again i am not demandig anything, if you have time and think of anything i am happy to test your thoughts.

shantigilbert, 2020-11-25:
Check `/emuelec/logs` it won't give you as detail information as the bios error check, but you can use the "missing bios" option in ES to check what BIOS you are missing and `emuelec.lo`g for all the other errors.

greekuser, 2020-11-26:
Ok, the next time i booted EmuELEC, from the welcome screen the resolution was mixed up. I hadn't done any change before shutdown. The only thing i done was to remove the sd card(with the EmuELEC) and boot from nand to Android to watch a movie. My tvbox (minix) has a button on it. You press it once to boot either from nand(android), or if a usb/sd card is inserted it autoboots from that(with only one press on the button). 
I have done many tests after that. 3.9 EmuELEC on an sd card, and 3.8 on a usb. I tested both. Both of them had messed up resolutions from the beginning from that point. And sdcard (3.9) wasn't even booted before to make any change. Last time worked ok (i sshed only 3.8 for testing purposes). So EmuELEC must be affected by Android. It seems that your theory of android's resolution is correct. I tried first adaptive resolution setting(auto change), and different colour depths settings in android with no success. When i changed only to 1080p in android, both 3.8 and 3.9 booted ok. But its strange because i am sure that i have booted EmuELEC normally when android was at 4k. So i repeated the first first step. 1.Set 4k in android, 2.Shutdown, 3.Sdcard in, 4.HOLD(not press) the button on the box to boot from a theoretical new partition, 5.EmuELEC booted fine(no messed resolution), 6.Shutdown, 7.Press(not hold now) the button to boot again, 8.EmuELEC booted fine again, 9.Shutdown, 10.Remove sdcard, 11.Press button to boot, 12.Android boots and shutdown without changing anything, 13.Insert sdcard, 14.Press button to boot, 15.EmueELEC boots with messed resolution, 16.Shutdown, 17.Hold button to boot, 18.EmuELEC boots fine.
From the above steps it's clear that if the last boot was android, it's resolution(4k) is stored somewhere and confuses EmuELEC. But if you HOLD the button to boot, somehow that holding bypasses that stored settings and EmuELEC boots fine. Also it seems that it overwrites that stored setting with it's own and the next time booting with only a press on the button is fine, as long as you don't boot android where you have to boot EmuELEC by holding the button again. All this if you have a 4k set-up of course. At 1080p there is no problem.
This behaviour is totally different from the bug at the error log(it also happen at 1080p). 
Finally, I SSHed and 3.9 to bypass the error and all is perfect now.....until i find something else, i hope i don't.

shantigilbert, 2020-11-26:
Read this: https://github.com/EmuELEC/EmuELEC/issues/355#issuecomment-733144430

Android resolution is changing Uboot settings (sets 4k resolution), when you press the button you clear those settings and EmuELEC runs fine.

greekuser, 2020-11-26:
I understand now the clear uboot settings thing, with the exception the minix's button is power button(not a reset button/hole) and you emulate a reset by holding it. It all makes sense now.

greekuser, 2020-11-26:
Shantigilbert thank you very much! Even if its not a real fix, bypassing/hidding the bug is an acceptable solution for me. If you dont have any other thoughts, feel free to close the issue as solved!
ps. I am curious to know.  You say that 4.0 will be 64bit only so s912 will be dropped. But cortex-a53 is advertised as 64bit. Or you will drop it because of the graphics drivers? Anyways, as i said before i dont know about that stuff , just a dumb question.

shantigilbert, 2020-11-26:
there are no graphic drivers for aarch64 that work in s912 (that I know of)

closed 2020-11-26 by shantigilbert

pepepaco, 2021-05-28:
instead of exit 0 at the line 5 add this at the end

echo "" > /tmp/es-restart && killall emulationstation

I saw this command in retropie forums.
[retropie forum](https://retropie.org.uk/forum/topic/8259/restart-es-by-commandline/8?_=1622188444504&lang=en-US)

shantigilbert, 2021-05-28:
@pepepaco that's a terrible idea, you are restarting ES needlessly, this is not retropie, and the error is not with ES.

shantigilbert, 2021-05-28:
Not if you followed the suggestion posted above, the issue is with the crappy GPU drivers of the S912 and the Terminal Emulator, the fix is to avoid the terminal emulator, which the suggestions does. But in if this works for you and you don't mind restarting ES then whatever works for you I guess.

greekuser, 2023-03-13:
Hi again. So after 2 years I am still rocking version 3.9. I have to ask though since amlogic s912 is indeed aarch64, if version 4.6 or any other has now support for s912 GPU?