EmuELEC/EmuELEC #358: How do you get Duke Nukem 3D to work?

Cee123, 2020-12-02:
Do you place the files for Duke Nukem 3D in /roms/ports/duke3d or /roms/ports/eduke32? Because I've tried both and neither of them are working.

I've checked the wiki page, nothing mentioned there.

shantigilbert, 2020-12-02:
First. Please fill out the Issue Template
Second,  Eduke is not included in 32bit builds. so if you are using v3.9 or below is not there.

Cee123, 2020-12-02:
> 
> 
> First. Please fill out the Issue Template
> Second, Eduke is not included in 32bit builds. so if you are using v3.9 or below is not there.

Oh ok, no worries. I'll do that then. Thanks. Yeah you're right, I'm using the unofficial test release. My bad, I should've notice it wasn't in the official release just yet.

shantigilbert, 2020-12-02:
use the missing bios screen and it will give you the correct path, it should be in /storage/roms/ports/....

Cee123, 2020-12-02:
> 
> 
> use the missing bios screen and it will give you the correct path, it should be in /storage/roms/ports/....

Alright cool, thanks for that. I'll try that and see where it says to place those files.

shantigilbert, 2020-12-02:
I just checked and its `/storage/roms/ports/eduke/duke3d.grp`

Cee123, 2020-12-02:
> I just checked and its `/storage/roms/ports/eduke/duke3d.grp`

Aha! Thank you! So that's what i did wrong. I was thinking the folder name was eduke32 like on retroarena. I'll try that should work.

closed 2020-12-02 by shantigilbert

Cee123, 2020-12-02:
Doesn't work just comes up with a general error message.

[image]

Cee123, 2020-12-02:
@shantigilbert Doesn't seem to work.

shantigilbert, 2020-12-02:
Open `/emuelec/ports/Duke Nukem.sh` and edit/add these lines

```
PORT="eduke32"
# init_port binary audio(alsa. pulseaudio, default)
init_port ${PORT} alsa
cd /storage/roms/ports/eduke
${PORT}
```

Cee123, 2020-12-02:
@shantigilbert Thanks for that! That worked! Those lines weren't in the script file. Thanks for taking the time out to help, really appreciate it. Cheers.