EmuELEC/EmuELEC #359: Display bug with Duke Nukem 3D on Odroid Go Advance

Cee123, 2020-12-04:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
Incorrect resolution or display of Duke Nukem 3D on the Odroid Go Advance. Multiple overlapping displays of the game across the top third of the screen.

**To Reproduce**
Steps to reproduce the behavior:
- Using the 4.0 test version of EmuElec.
- Under ./config/emuelec/ports/Duke Nukem.sh (if you do not have these lines in the script file) add these:
PORT="eduke32"
# init_port binary audio(alsa. pulseaudio, default)
init_port ${PORT} alsa
cd /storage/roms/ports/eduke
${PORT}

**Expected behavior**
Should display fullscreen.

**Screenshots**
[image]

**EmuELEC version:**
 - Version 4.0 Test

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
Odroid Go Advance.

**Additional context**
Tried changing the resolution under the config files as well as the screen size, but it doesn't make any difference. I'm not sure if anyone is aware of this bug, so I just thought I'd report it to you guys here. This only seems to happen on the OGA, on everything else it runs perfectly.

shantigilbert, 2020-12-07:
I don't think I have tested Eduke on OGA, so I am not sure why this is happening, but seems resolution related. Try modifying `/storage/.config/eduke32/eduke32.cfg` and change/add these lines under `[Screen Setup]`: 

```
ScreenHeight = 320
ScreenWidth = 480
```

Cee123, 2020-12-07:
Aha! You were right. Thanks! That worked! changing the screenheight and screenwidth worked. I tried other resolutions before but for some reason none of them stuck.

Cee123, 2020-12-07:
That seemed to have fixed it.

closed 2020-12-07 by Cee123

johannesbehr, 2021-04-05:
Could you please share your complete eduke32.cfg for the OGA?
I'm not getting the corect resolution to work :(

Cee123, 2021-04-06:
> 
> 
> Could you please share your complete eduke32.cfg for the OGA?
> I'm not getting the corect resolution to work :(

Yeah of course. There are two separate config files on two separate partitions. Both called eduke32.cfg.

The first one goes in ./config/eduke32 on your storage partition. And the other one goes in the /ports/duke3d on your EEROMS partition. Hope it helps.

[Storage (Linux Partition).zip](https://github.com/EmuELEC/EmuELEC/files/6263166/Storage.Linux.Partition.zip)
[EERoms (Windows Partition).zip](https://github.com/EmuELEC/EmuELEC/files/6263167/EERoms.Windows.Partition.zip)

Weestuarty, 2022-06-28:
Hey i keep getting stuck on a black screen could you confirm for me what DN3d files you added to the base install and anything else. desperately need help