EmuELEC/EmuELEC #360: Installation issues on UGOOS X3 Plus

RedEye80, 2020-12-05:
Hi there, just joined Github and attempting to use EmuELEC for the first time. I have a UGOOS X3 Plus i.e. s905x3 4gb updated to latest firmware 0.37 by the manufacturer.

So I flash my 128gb micro sd with etcher, move the appropriate sm1_s905x3_4g.dtb file to the root of the sd, rename it to dtb.img and it all starts loading fine until I get to the  screen where it should detect my controllers. Nothing works here from a variety of gamepads including my flydigi Apex to generic wired PS3 controller. Nothing works here including remotes or airmice. I tried using different files too. Maybe not the right one? By the way this regards release 3.9 and I downloaded EmuELEC-Amlogic-ng.arm-3.9-Generic.img.gz from here. Is the usb on the box not supported? Feel like I'm so close no matter how far...

Any help will be greatly appreciated. Thank you in advance

AlienatedDiego, 2020-12-05:
try other s905x3 .dtb; per your device specs sm1\_s905x3\_4g\_1gbit.dtb seems appropriate.

RedEye80, 2020-12-05:
Thank you Diego but I tried and tried again different *.dtb and no joy. It seems like EMUELEC does not recognize and power up the usb ports at all. See when I plug in a mouse the red light does not come on and the flydigi does not pair up and switches off after a while although it pairs up straight away even before android 9 has booted up on the same box. No lights come on the wired gamepads neither. Unless someone suggests a better solution I'm giving up...which is a big shame because I feel like I got so close. 

However..... If I was to purchase another box right now what would you guys suggest for good performance on Emuelec?

AlienatedDiego, 2020-12-05:
actually any Amlogic soc performs well in EmuELEC, but if it can afford it s922x is the best (beelink gt king / pro, Odroid N2 / N2 +)

AlienatedDiego, 2020-12-05:
A colleague from the EmuELEC team @samwilson with more experience in tvbox comments / provides information from CoreELEC forums regarding another Ugoos model with the same symptoms, you may be interested. It seems that there is no specific .dtb for it BUT .... you can try those commands using ssh and see if the usb problem is solved, at least temporarily before trying to modify any files. 

https://discourse.coreelec.org/t/x2-cube-usb-ports-do-not-work/7679

AlienatedDiego, 2020-12-05:
this guide can be useful to try to link bluetooth controllers https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth

RedEye80, 2020-12-06:
Thanks again for trying to solve my problem. It seems like the discourge link you posted is not working however. Would be interesting to see because If the other user managed to do it I might still have a chance with EmuELEC's current release. It seems like the usb ports of the specific device are not working at all in EmuELEC and the cube is very similar to mine.

In regards to Bluetooth, automatic mode did not work so will try my luck with putty and commands to see if I get anywhere.

RedEye80, 2020-12-07:
So just a quick update for my situation:

I tried Bluetooth automatic mode as described in wiki with no success.
Then the link provided by Diego https://discourse.coreelec.org/t/x2-cube-usb-ports-do-not-work/7679 finally worked. Website must have been down when I checked.

So as suggested to another user with a similar model I added the following lines to the custom_start.sh file using windows notepad and managed to get one of the usb ports working.

echo 432 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio432/direction
echo 1 >  /sys/class/gpio/gpio432/value
echo 432 > /sys/class/gpio/unexport

And I managed to get a wired controller identified and go past that screen and enter EmuELEC. So the adventure begins now but I feel like I managed to get past a major bottleneck for a newb like me anyway. Will try to get more controllers working including the Bluetooth and compile a good game library.

Looking forward to compare performance of emuELEC with say DIG and retroarch running on Android.

Diego I really appreciate your help. Could not have done it without it.

I don't know how popular UGOOS X series is with emuELEC but if it is a persistent problem, do they maybe need to have their own separate SM1 file in device trees in a newer version? Thanks again Diego!

shantigilbert, 2020-12-09:
So is this considered fixed?

RedEye80, 2020-12-09:
Yup! Happy as Larry here and nice work by the way. Loving EmuELEC! Don't think I'll be playing any of the supported emulators on android mode or pc anymore.

closed 2020-12-09 by RedEye80