EmuELEC/EmuELEC #363: .sav file could be generated

ggpaue, 2020-12-06:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
On my anbernic rg351p device, gba game saving .sav file could not be generated
**To Reproduce**
Steps to reproduce the behavior:
1. Open a gba game on device, for example pokemon
2. play for a while
3. save
4. quit the game
5. No .sav file generated, so the game could not be saved

**Expected behavior**
a .sav file should be generated and mark as save file
**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version v3.7

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: [e.g. S095x3]
 - Brand: Anbernic
 - Model rg351p
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

ggpaue, 2020-12-06:
just notice rg351p is not supported because they break the copyright, will close this issue, but anyone welcome to comment

closed 2020-12-06 by ggpaue

shantigilbert, 2020-12-06:
https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices