EmuELEC/EmuELEC #364: Time synchronized error

stats121, 2020-12-07:
**Describe the bug**
Time synchronized error appears frequently between load screens in Odroid Go Advance on v3.9

**To Reproduce**
Happens at start-up or when loading an Emulator

**Expected behavior**
No failure

**Screenshots**
[image]

**EmuELEC version:**
v3.9

shantigilbert, 2020-12-07:
That is really not a bug, Timesync needs internet to work correctly , it is annoying tho, and I will try to find another way to do it, but for the time being, there is nothing to fix.

hardknoxni69a, 2021-01-24:
how is this not considered high priority when it effectively freezes our system forcing a hard reboot on our SD card? I may be missing something but how am I supposed to restart when this kills the system?

shantigilbert, 2021-01-24:
@hardknoxni69a whatever it is that is causing your freezes is not related to this, just because you see it on screen it does not mean it's the issue

hardknoxni69a, 2021-01-25:
It is a fresh flash so not sure what would be causing the issue. But ya the only thing I see on screen is the timer error. How do I fix the time

shantigilbert, 2021-02-14:
Fixed in v4.0

closed 2021-02-14 by shantigilbert