EmuELEC/EmuELEC #368: PS3 controller issue

Cebyo, 2020-12-15:
Hello,
I have a controller problem. The first time I installed emuelec I connected a logitech f710. After that, I wanted to connect 2 ps3 controllers and they are recognized in the menu to navigate between consoles and roms but as soon as I launch a game, retroarch tells me that the ps3 controller is not configured. I can reconnect the logitech controller to configure it in the "input" menu but anyway, as soon as I quit I have changes that I made are not saved and I keep coming back to the same problem.

shantigilbert, 2020-12-15:
Configure them in Emulationstation, not in retroarch

closed 2020-12-15 by shantigilbert

Cebyo, 2020-12-15:
The 2 controllers are configured under emulationstation via the menu and both work fine in the "es" menu but when I run a game the controller doesn't work anymore, I can't even open the retroarch menu to quit.

So I tried to go directly into retroarch thanks to the quit menu of "es" and there it tells me that the ps3 controller is not configured.

shantigilbert, 2020-12-15:
Then try saving the profile in retroarch before quiting

shantigilbert, 2020-12-15:
Using the other controller to open the settings, and configure them, then save the profile

Cebyo, 2020-12-15:
here it works. In fact, the first remark you made was the correct one. I disconnected the controller f710 and I reconfigured the 2 controllers ps3 under ES and it works. I probably did it wrong the first time, or keeping the logitech connected, I don't know .... But anyway thank you.