EmuELEC/EmuELEC #369: Requirement that special chars in WiFi Password must be escaped is unclear in GUI

brnl, 2020-12-17:
**Is your feature request related to a problem? Please describe.**
I've been struggling a lot today to get my RG351P on my WiFi connection. The problem turned out to be that I had a $-sign in the password that had to be escaped! I only found out after connecting a wired network connection to configure WiFi via SSH. In the configuration file, it is written that special chars should be escaped but in the GUI it's not clear.

Also: _Connection errors are not reported back to the user._

**Describe the solution you'd like**
When the WiFi can't connect, you only get a "not connected" status.
Solution: Display an error that the password was incorrect. In this message, you can point the user on the fact that special chars should be escaped.

**Describe alternatives you've considered**
Display a notification if a user enters an special char or display that message natively.

**Additional context**
When creating this issue, I found that it is written in the Wiki [here](https://github.com/EmuELEC/EmuELEC/wiki/WiFI#important-all-versions), but I would really appreciate it if it was more clear in the GUI.

I think with one of these additions it will be easier for other users with complex passwords to fix their issue quicker than I did. I didn't have much luck with my Google-Fu eighter I must add!

shantigilbert, 2020-12-17:
Thanks for the input. Not really interested in fixing this, the wiki should be more than enough for this very minor issue. Please note that although this affects all devices the rg351p is not officially supported.

closed 2020-12-17 by shantigilbert

brnl, 2020-12-17:
Fine, whatever. But your comment about the rg351p not being supported is, as you say yourself totally irrelevant.

shantigilbert, 2020-12-17:
It is very relevant, I do not want to support those devices that do not care about licenses or copyright or even giving support, that is my choice, you don't have to agree. If you want to post it on 351ELEC then by all means do so. But please don't ask for support here for that device (or any of those others), it is clearly stated.

brnl, 2020-12-17:
I didn't ask for support, I offered an idea to make your project better to new users. But whatever, you do you. Have a great day.

shantigilbert, 2020-12-17:
Ask anbernic, I am sure they will help :)

brnl, 2020-12-17:
And about the licenses: How could I know? I'm really sorry about that and understand that must be frustrating but don't blame that on me, please. I was just enthusiastic... :-(