EmuELEC/EmuELEC #371: How to get xbox one controller to work over bluetooth

wargex, 2020-12-17:
I don't know where you would like me to post this. 
But I figured out that if you quit EmuELEC, and load in to RetroArch set your xbox one joysticks there. 
Quit RetroArch load back in to EmuELEC, reset your controller mappings again in EmuELEC.. 
EmuELEC just finds them. 
It's that easy, I mean, more steps then setting it, but it works. 
I'm not sure if this is known already, but I haven't seen anyone say anything and if it is known, its just not in the wiki, in the wiki it just says it doesn't work. But it does. 
But I hope this helps someone out there. I headed here as soon as I figured it out.

________________________________________________________________________________________
Also kinda separate, could not get OG or new xbox one controllers to work over usb. (not new, new, it's not a series x controller if they're even any different)
________________________________________________________________________________________
Also for some reason EmuELEC remaps the little android keyboard remotes once its in game. Both directions and the other buttons[abxy] mapped to the same directional buttons. It's one of those generic 10-15$ wireless things. Unbranded Rii remote thing, you can find it on amazon if you're unsure of what I mean. I only bring it up cause the mappings work fine for the most part in EmuELEC home they just don't stay the same in game.

AlienatedDiego, 2020-12-17:
hi, thanks for the report. Yes, we know about this procedure but the reason why it is not published is because it is not functional in all XO V2 controllers, only a small percentage seems to link correctly, we do not know the specific reason, different version bt, firm, anything.

You can now save your controller's .conf file that you will find in the "joypads" folder to add it to later versions of emuELEC without repeating the procedure each time.

AlienatedDiego, 2020-12-17:
yes, usb keyboards (also RII) will be used to navigate in Emulation Station and for emulators that support the use of keyboard.

the steam controller is not currently supported.

Djglitch2011, 2020-12-24:
> Post updates are mostly fixing grammar/being more specific in wording.
> 
> I don't know where you would like me to post this.
> But I figured out that if you quit EmuELEC, and load in to RetroArch set your xbox one joysticks & controls there.
> Quit RetroArch load back in to EmuELEC, reset your controller mappings again in EmuELEC..
> EmuELEC just finds them.
> It's that easy, I mean, more steps than just setting it, but it works.
> I'm not sure if this is known already, but I haven't seen anyone say anything and if it is known, its just not in the wiki, in the wiki it just says it doesn't work. But it does.
> But I hope this helps someone out there. I headed here as soon as I figured it out.
> 
> Also kinda separate, could not get OG or new xbox one controllers to work over usb. (not new, new, it's not a series x controller if they're even any different)
> 
> Also for some reason EmuELEC remaps the little android keyboard remotes once its in game. Both directions and the other buttons[abxy] mapped to the same directional buttons. It's one of those generic 10-15$ wireless things. Unbranded Rii remote thing, you can find it on amazon if you're unsure of what I mean. I only bring it up cause the mappings work fine for the most part in EmuELEC home they just don't stay the same in game.
> 
> Steam controller did not work for me over dongle

Это действительно сработало спасибо, теперь можно без донгла использовать

shantigilbert, 2020-12-25:
> Also kinda separate, could not get OG or new xbox one controllers to work over usb. (not new, new, it's not a series x controller if they're even any different)

OG as in Xbox OG or do you mean Xbox one controllers? I tested Xbox OG (the real "1") and they worked fine, Xbox 360 as well. the problems are only with Xbox one and up, and its weird because some of them work some of them do not. I have no idea of why this is. 

As for the keyboard, this might be the configuration script, but since I have no mini keyboard like that I cannot test.

wargex, 2020-12-27:
> OG as in Xbox OG or do you mean Xbox one controllers? I tested Xbox OG (the real "1") and they worked fine, Xbox 360 as well. the problems are only with Xbox one and up, and its weird because some [...]

Nah I mean OG Xbox One controller, after Xbox One S came out controllers then had 3.5 aux, and bluetooth. The controllers made at launch did not have aux or bluetooth. The OG Xbox One controller didn't connect through USB, is what I mean. You can call them V1 controllers idk if there's like a community standard on that or whatever, but they're different controllers. I hear the new Series controllers are different too, so I think there are at least 3 different official xbox one controllers.

closed 2022-01-21 by shantigilbert