EmuELEC/EmuELEC #372: Has it been tested on the S905x4 yet?

SecularSteve, 2020-12-20:
The Verizon Stream TV just came out, and it's one of the first to utilise the brand new Amlogic S905x4.

https://www.youtube.com/watch?v=n_ItkYemb98

Has there been an attempt to test this (or a similar) device yet?

shantigilbert, 2020-12-20:
I do not have any device with that chipset, so I assume it does not work.

SecularSteve, 2021-01-20:
**New Device**

https://www.youtube.com/watch?v=k-Hp1h_2GJM

dm2912, 2021-01-21:
There are plenty of new devices but if nobody has one there is no way of knowing. 

Your best place to check is the coreelec forums to see if there is a dtb file for the particular box. If there is no dtb, its not going to work..

closed 2021-04-24 by shantigilbert

THEONE12345678, 2021-08-13:
Who tests it on s905x4?
 4.2 is not working.