EmuELEC/EmuELEC #374: How to take a screenshot in OGA on terminal?

mtoensing, 2020-12-23:
Hi! 

Normally I take screenshots with https://github.com/AndrewFromMelbourne/raspi2png but I am unable to unstall this on EmuELEC. How do you guys take direkt screenshots on the terminal?

shantigilbert, 2020-12-23:
There is currently no option that I know of, you could try dumping the whole fb to a png, but at the moment I do not remember the exact command

I will look into the program you linked to see if I can incorporate it in future versions.

mtoensing, 2020-12-23:
That would be great. I am working on an article about the best images for the RockChip RK3326 handhelds. But I am not able to take high quality screenshots in EmuELEC... =/

shantigilbert, 2020-12-23:
Is this functionality available in other distros for the Rk3326 ?

mtoensing, 2020-12-23:
If you have an open OS you can install software. EmuELEC uses OpenELEC which seems to be designed around a locked filesystem.

shantigilbert, 2020-12-23:
Sure, but my question was if this is already available in other distros, or do you mean you installed that software you linked? 

if that is the case, try installing the binary to `/emuelec/bin` and libs to `/emuelec/lib`

mtoensing, 2020-12-23:
I installed the software. 
If I try to do this in EmuELEC:

`curl -sL https://raw.githubusercontent.com/AndrewFromMelbourne/raspi2png/master/installer.sh | bash -`

I only get errors. 

If I try this: 

`apt-get -y install libpng12-dev git-core`

It tells me this is not the way it works on OpenELEC. How do I install it in these folders?

mtoensing, 2020-12-23:
Even git clone https://github.com/AndrewFromMelbourne/raspi2png.git does not work. I am not a linux wizard so I think I might stop here.

shantigilbert, 2020-12-23:
Yeah you won't be able to do that, EmuELEC is not a typical OS, if you have the binary already compiled you can follow what I said earlier, else, it will need to be compiled elsewhere first, that is why I asked if it was available on other Rk3326 devices already.

shantigilbert, 2020-12-23:
I can check this in a few hours and let you know if it even works :)

mtoensing, 2020-12-23:
Wow! Thank you!

shantigilbert, 2020-12-24:
[screenshot.zip](https://github.com/EmuELEC/EmuELEC/files/5738662/screenshot.zip)
I compiled a version of fbgrab, but since I no longer have a 32bit environment I was not able to test it. 

Can you unzip it into ConfigFiles/emuelec/bin (using network share) and in SSH type `screenshot.sh` a screenshot of whats on the FB should appear in `/storage/roms/screenshots` 

I will repeat that I have not been able to test this, so please let me know if it worked.

mtoensing, 2020-12-24:
Wie! Thank you! I will try and respond asap

mtoensing, 2020-12-24:
Hi, 

thank you very much. It seems to work ... in parts. It takes a screenshot and saves it. See attached screenshot. But this from the boot process and not what the oga is currently displaying. I see the menu and browsed the roms and made several screenshots. All look like the attached one. 

[image]

shantigilbert, 2020-12-24:
[screenshot_2.zip](https://github.com/EmuELEC/EmuELEC/files/5740364/screenshot_2.zip)

Can you try with this one? same procedure

mtoensing, 2020-12-24:
Does this answer your question? Thank you! 
[image]

closed 2020-12-24 by mtoensing

mtoensing, 2020-12-29:
@shantigilbert 

One thing that could be changed: 

SCREENDIR="/storage/screenshots" 

this is better than /storage/roms/screenshots because there is already a directory for that.

mtoensing, 2020-12-29:
Here is the new name: https://github.com/mtoensing/es-theme-Graphite-OGA

Sorry for the inconvenience.