EmuELEC/EmuELEC #376: Shaders GLSL not work on X96 mini 2/16 (s905w).

bennington111, 2020-12-24:
Emuelec install on X96 mini 2/16 (**s905w**).
**Shaders GLSL not work**. Nes, Snes, Megadrive games. Failed to apply shader preset. How to fix this bug?

zensand, 2021-01-25:
He, I would like to ask you how did you get the x96mini to reboot? Because mine sticks at the begin logo. And i have tried every device tree (x96 mini 2gig ram & 16 gig storage with andriod 7). and even tried different sd cards... hope you can help me out ;-)

bennington111, 2021-01-25:
> He, I would like to ask you how did you get the x96mini to reboot? Because mine sticks at the begin logo. And i have tried every device tree (x96 mini 2gig ram & 16 gig storage with andriod 7). and even tried different sd cards... hope you can help me out ;-)

I press reset button, then connect power cable. I use **gxl_p281_2g.dtb** and use **ATVxperience 4.2.1** on Android 9.

closed 2021-04-01 by shantigilbert