EmuELEC/EmuELEC #377: Problem with n64 roms

mike743232, 2020-12-24:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**

after upgrade from 3.5 to 3.9 it does not load n64 roms

**To Reproduce**
Steps to reproduce the behavior:

usual steps when it was 3.5 and they worked

**Expected behavior**

that roms work

**Screenshots**
impossibile

**EmuELEC version:**
3.9

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: s905m
 - Brand: i dont know
 - Model super console x
 - Version i dont know

**Additional context**

apart from this problem already from 3.5 roms with .crt extension on c64 do not work. To make them work I had to convert them to zip

SamuEDL, 2020-12-24:
Good afternoon and Merry Christmas.
We don't support super console x, if you are upgrading from the modified emuelec that comes with it. It is possibly what is giving you failures.

mike743232, 2020-12-25:
No emuelec is not modified, and the one taken from github, cmq if you can help me I saw that the roms do not start and from this error "error init libretro symbols ()"

shantigilbert, 2020-12-25:
> No emuelec is not modified, and the one taken from github, cmq if you can help me I saw that the roms do not start and from this error "error init libretro symbols ()"

You seem to have miss-understood, the one in the console x is modified, we do not support it. Either do a clean install, and if you have the same issue we can help otherwise ask them for support. 

also please read the template, it clearly says to delete all the text above certain line, so that it shows that you have read the rules, clearly you have not. 

Closing this issue, feel free to reopen it after you have done a clean install and if you still have the issue

closed 2020-12-25 by shantigilbert