EmuELEC/EmuELEC #378: N64 emulators freeze after a few minutes of playing. (Ram memory full)

durstboiler, 2020-12-25:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
The N64 emulators freeze after a few minutes of play due to the infinite filling of the ram memory.

**EmuELEC version:**
 - Version 3.9

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: s905x
 - Model x96 1gb 8gb

**Additional context**
The duration varies from one game to another, with some games you can play 30min, others freeze after 1 min in the menus.
Each time the cause is the same, the ram memory fills up, (as if the textures accumulated without there ever being a dump), then at 690mb (on 806mb), freeze.
So of course, my box has very little ram, but the same problem would happen on any other system, it would just take longer.

shantigilbert, 2020-12-25:
First, please provide LOGS, second please read the template, it clearly says to delete all the text above certain line, so that it shows that you have read the rules, clearly you have not, and third, 1gb ram is not really recommended for EmuELEC.

durstboiler, 2020-12-25:
I do not know how to provide a log file, The box freezes does not crash, so I have to turn it off and no LOG file is created. Otherwise I shut down the game just before the crash by checking the ram, but I don't know if this will teach you anything.

`EmuELEC Run Log
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 3.9 (Amlogic.arm)
PLATFORM: n64
ROM NAME: /storage/roms/n64/Lylat Wars (Europe) (En,Fr,De).n64
BASE ROM NAME: Lylat Wars (Europe) (En,Fr,De).n64
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/n64/Lylat Wars (Europe) (En,Fr,De).n64
2nd Argument: -Pn64
3rd Argument: --core=mupen64plus
4th Argument: --emulator=libretro
Full arguments: /storage/roms/n64/Lylat Wars (Europe) (En,Fr,De).n64 -Pn64 --core=mupen64plus --emulator=libretro --controllers=-p1index 0 -p1guid 030000006d04000018c2000010010000 
Run Command is:
/usr/bin/retroarch -v -L /tmp/cores/mupen64plus_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/n64/Lylat Wars (Europe) (En,Fr,De).n64
Emulator Output is:
[INFO] RetroArch 1.9.0 (Git ddef65f626)
[INFO] [Overrides]: Redirection du fichier de sauvegarde vers "/storage/.config/retroarch/saves/Lylat Wars (Europe) (En,Fr,De).srm".
[INFO] [Overrides]: Redirection du fichier de sauvegarde instantanée vers "/storage/roms/n64/Lylat Wars (Europe) (En,Fr,De).state".
[INFO] === Build =======================================
[INFO] Capacités:  NEON VFPv3 VFPv4 ASIMD
[INFO] Built: Oct 23 2020
[INFO] Version: 1.9.0
[INFO] Git: ddef65f626
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [CORE]: Loading dynamic libretro core from: "/tmp/cores/mupen64plus_libretro.so"
[INFO] [Overrides]: Core-specific overrides found at /storage/.config/retroarch/config/Mupen64Plus GLES2/Mupen64Plus GLES2.cfg.
[INFO] [Overrides]: No content-dir-specific overrides found at /storage/.config/retroarch/config/Mupen64Plus GLES2/n64.cfg.
[INFO] [Overrides]: No game-specific overrides found at /storage/.config/retroarch/config/Mupen64Plus GLES2/Lylat Wars (Europe) (En,Fr,De).cfg.
[INFO] Config: appending config "/storage/.config/retroarch/config/Mupen64Plus GLES2/Mupen64Plus GLES2.cfg"
[INFO] [Environ]: SET_VARIABLES.
[INFO] Options par jeu : options de cœur spécifiques au jeu trouvées dans /storage/.config/retroarch/config/Mupen64Plus GLES2/Lylat Wars (Europe) (En,Fr,De).opt
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Remaps]: remap directory: /storage/.config/retroarch/config/remappings
[INFO] [Remaps]: core-specific remap found at /storage/.config/retroarch/config/remappings/Mupen64Plus GLES2/Mupen64Plus GLES2.rmp.
[INFO] [Overrides]: Redirection du fichier de sauvegarde vers "/storage/.config/retroarch/saves/Mupen64Plus GLES2/Lylat Wars (Europe) (En,Fr,De).srm".
[INFO] [Overrides]: Redirection du fichier de sauvegarde instantanée vers "/storage/.config/retroarch/states/Mupen64Plus/Lylat Wars (Europe) (En,Fr,De).state".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/storage/roms/bios".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/storage/roms/bios".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [CONTENT LOAD]: Chargement du fichier de contenu: /storage/roms/n64/Lylat Wars (Europe) (En,Fr,De).n64.
[INFO] Impossible de trouver un patch valide pour ce contenu.
[INFO] [Environ]: SET_HW_RENDER.
[INFO] Requesting OpenGLES2 context.
[INFO] [Environ]: SYSTEM_DIRECTORY: "/storage/roms/bios".
[libretro INFO] EmuThread: M64CMD_ROM_OPEN
[libretro INFO] mupen64plus: Goodname: Lylat Wars (E) (M3) [!]
[libretro INFO] mupen64plus: Name: STARFOX64           
[libretro INFO] mupen64plus: MD5: 884CCCA35CBEEDB8ED288326F9662100
[libretro INFO] mupen64plus: CRC: F4CBE92C B392ED12
[libretro INFO] mupen64plus: Imagetype: .v64 (byteswapped)
[libretro INFO] mupen64plus: Rom size: 12582912 bytes (or 12 Mb or 96 Megabits)
[libretro INFO] mupen64plus: ClockRate = F
[libretro INFO] mupen64plus: Version: 1448
[libretro INFO] mupen64plus: Manufacturer: Nintendo
[libretro INFO] mupen64plus: Cartridge_ID: 5846
[libretro INFO] mupen64plus: Country: Europe (0x50)
[libretro INFO] mupen64plus: PC = 80000400
[libretro INFO] mupen64plus: Save type: 0
[libretro INFO] EmuThread: M64CMD_ROM_GET_HEADER
[INFO] [SRAM]: Chargement de la SRAM ignoré..
[... 122 lines cut ...]
[INFO] [GL]: VSync => off
[INFO] [EGL]: eglSwapInterval(0)
[INFO] [GL]: VSync => on
[INFO] [EGL]: eglSwapInterval(1)
[INFO] [GL]: VSync => off
[INFO] [EGL]: eglSwapInterval(0)
[INFO] [GL]: VSync => on
[INFO] [EGL]: eglSwapInterval(1)
[INFO] Sauvegarde instantanée en cours: "/storage/.config/retroarch/states/Mupen64Plus/Lylat Wars (Europe) (En,Fr,De).state.auto", Taille de la sauvegarde instantanée: 16789316 octets.
[INFO] Sauvegarde instantanée automatique vers "/storage/.config/retroarch/states/Mupen64Plus/Lylat Wars (Europe) (En,Fr,De).state.auto" succeeded.
[INFO] [Overrides]: Configuration overrides unloaded, original configuration restored.
[INFO] [config] Nouvelle configuration enregistrée vers "/storage/.config/retroarch/retroarch.cfg".
[... cut ...]
[INFO] [SRAM]: Enregistré avec succès vers "/storage/.config/retroarch/saves/Mupen64Plus GLES2/Lylat Wars (Europe) (En,Fr,De).srm".
[INFO] Content ran for a total of: 00 hours, 00 minutes, 21 seconds.
[INFO] [CORE]: Unloading game..
[INFO] [CORE]: Unloading core..
[libretro INFO] mupen64plus: Stopping emulation.
[INFO] [PERF]: Performance counters (libretro):
[INFO] [CORE]: Unloading core symbols..
[INFO] [Core Options]: Saved game-specific core options to "/storage/.config/retroarch/config/Mupen64Plus GLES2/Lylat Wars (Europe) (En,Fr,De).opt"
[?12l[?25hcontext_reset.
plugin_start_gfx success.
exit 0`

dm2912, 2020-12-25:
S905x with 1gb ram is really not enough to run N64 games

My 905w (same chip) would not handle it. My 905x3 (4gb ram) did but even that had struggles with some games.

durstboiler, 2020-12-25:
Another very strange thing about the operation of the ram: in Mario Golf, by doing nothing other than turning on myself, the ram increases until the box freezes, as if the same textures were loaded indefinitely.

shantigilbert, 2020-12-25:
> the management of the ram of the emulator.

If this is the case the issue should be opened in the github of the core, as we cannot fix such big problem. Or try another core

durstboiler, 2020-12-25:
This is the same for all n64 cores, the only thing that slows down the ram filling is not using texture enhancement, but if I play for a long time it will inevitably happen.

shantigilbert, 2020-12-25:
The this is an issue that should be reported to the core directly, there is nothing we can do about this in general, all I know is that people with 2gb or 4gb do not have this issue. 1gb is still not enough for EmuELEC.

ricardoandren, 2020-12-26:
I have the same problem

durstboiler, 2020-12-26:
I am doing some tests.
Already, you should know that there are games, at certain times when the textures are loaded as in an infinite loop.
For example, the Starfox 64 map, or the Mario golf game phase.
To make things better, in mupen64plus-next:
-MAX TEXTURE CACHE SIZE : 1500
This caps the loading of large textures, as I understand it.

-TEXTURE ENHANCEMENT : AS IS or NONE
Exemple (After reaching the cap.) Mario Golf : 
NONE: 476mo 
AS IS: 490mo
HQ2X: 587mo
2XBRZ: 650mo

-RESOLUTION : ~15-20mo saving by lowering it.
-NOISE EMULATION : OFF
~15-20mo saving.

closed 2021-04-24 by shantigilbert